luc-github/Repetier-Firmware-4-Davinci #184: Bed and Extruder not heating from menu commands.

Chriss6181, 2016-09-01:
Hello I'm new and am not sure if this has been resolved, But I flashed my first generation Da Vinci 1.0 with the latest firmware and the bed and extruders does not heat up. Commands like auto bed level manual bed level and clean nozzle just sit while the screen displays heating. this was not an issue with .91 firmware or stock. Am I doing something wrong?

luc-github, 2016-09-01:
Did you load default settings / or did a M502/M500 command ?

Chriss6181, 2016-09-01:
Yes I did it still didn't work how ever I just refreshed and loaded the fail safes it took two trys but seems to be working. Thank you.

closed 2016-09-01 by Chriss6181

luc-github, 2016-09-01:
you need to wait 20s after saving to EEPROM or any change before power down printer - saving EEPROM from memory to SD is done every 15s