luc-github/Repetier-Firmware-4-Davinci #186: Scroll speed for SD-related menu ops

extrapilot, 2016-09-07:
Hi
Is there a code-based limiter for the page/scroll speed within the “Print File” menu? On my machine (a 1.0A, running a Toshiba Airware SD card, Repetier firmware 0.92), holding down the ‘down’ key to access a file in the file list seems to require about 4 second for the > indicator to move to the next line/filename. Likewise for the “Delete File” menu under the “SD CARD” menu. Navigation in the other sections of the menu is speedy, so it doesn’t seem to be related to a timer on the button press itself. Since the file names are already shown on the LCD- this doesn’t seem to be a ‘read next file’ thing on the SD card- it is more like a delay in moving through a filename buffer.

Any idea why this may be happening? Perhaps there is just some problem with the SD card, and the code is stuck in a wait state until the SD card returns data.

Regards

luc-github, 2016-09-07:
Hi no there is not 
I cannot reproduce on my 2.0 with flash air so let me ask:
do you have same issue with micro sd ?
what version of FW do you use ? the long one starting by 1. and ending by A in setting/system version
this happen after a start ? or after deleting a file ?
How many files/dir on your sd card root?

extrapilot, 2016-09-07:
Hi
Not sure what you mean by micro sd. So far as I know, there is only the standard SD slot on the controller board. Firmware reads: 1.15-10-30_1.1, I dont see an 'A' terminating the string, so perhaps it is off the screen with no word wrap?

I dont recall when this started. I believe it was after a firmware upgrade. I had to modify the thermistor tables several times, and I tended to download current firmware when that was required. I know this was not a problem initially.

Im not sure about the filesystem, as some info may be masked by Repetier, and I dont have a proper web interface on the FlashAir card (only upload). So far as I know, there are no directories, and just the few standard systems files, along with 2 gcode files. I have deleted the gcode and posted new files many times over the past month- and no change in behavior. 

It is entirely possible this is a FlashAir card problem, where the SPI read routine is probably just in a wait loop on reading. But I find it strange that both gcode files show on the screen when I access Print File, and just moving the > up or down can take 4 seconds... If it is just reading this from an array/structure, this shouldnt happen (since it isnt a problem in other submenus). Maybe there is a re-read thing happening with the SD in the background that is blocking...

It might make sense for me to do a reset on the SD card- perhaps it was corrupted during a powerdown etc. But hard to imagine- since it seems to work fine via ethernet-

Kind regards- and thanks so much for your work on this firmware; it makes for a very capable and affordable platform!

luc-github, 2016-09-07:
I talked about microsd because you wrote you have 1.0a , if you have normal SD you have 1.0 as the 1.15-10-30_1.1 says

about flash air I wrote a filemanager - you may try it https://github.com/luc-github/Flashair-FM
 
about the delay- would be good to check with another SDCard this will tell us if issue follow hardware or software

luc-github, 2016-09-11:
any update on this ?

luc-github, 2016-09-13:
I close issue if no update - please reopen if needed

closed 2016-09-13 by luc-github

extrapilot, 2016-10-13:
Sorry for the delay- I had RMAd the FlashAir card, as it was behaving abnormally in test. I wanted to check with that card type, due to its Fat32 requirement etc. And, problem resolved. The scrolling is now fast as can be expected. Strange in that I didnt see much in the way of other problems (failures to access SD card via SPI interface, etc). But, why ask why- if anyone sees a similar problem, well- this is at least one confirmed cause.

Regards