luc-github/Repetier-Firmware-4-Davinci #187: Error Compiling

hletchworth, 2016-09-08:
Hello,

I get the following error compiling, even after transferring the hardware folder. Any help is appreciated!

`Arduino: 1.5.8 (Windows 8), Board: "Arduino Uno"

WRITE(SDSS, HIGH);
        ^
HAL.h:147:111: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
HAL.h:670:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, HIGH);
        ^
HAL.h: In static member function 'static void HAL::spiSend(const uint8_t*, size_t)':
HAL.h:147:33: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
HAL.h:677:9: note: in expansion of macro 'WRITE'
         WRITE(SDSS, LOW);
         ^
HAL.h:147:111: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
HAL.h:677:9: note: in expansion of macro 'WRITE'
         WRITE(SDSS, LOW);
         ^
HAL.h:147:33: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
HAL.h:681:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, HIGH);
        ^
HAL.h:147:111: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
HAL.h:681:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, HIGH);
        ^
HAL.h: In static member function 'static void HAL::spiSendBlock(uint8_t, const uint8_t*)':
HAL.h:147:33: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
HAL.h:689:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, LOW);
        ^
HAL.h:147:111: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
HAL.h:689:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, LOW);
        ^
HAL.h:147:33: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
HAL.h:696:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, HIGH);
        ^
HAL.h:147:111: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
HAL.h:696:8: note: in expansion of macro 'WRITE'
        WRITE(SDSS, HIGH);
        ^
In file included from Repetier.h:370:0,
                 from Commands.cpp:22:
HAL.h: In static member function 'static void HAL::startWatchdog()':
HAL.h:744:13: error: 'WDT_MR_WDRSTEN' was not declared in this scope
   timeout = WDT_MR_WDRSTEN | WDT_MR_WDV(timeout) | WDT_MR_WDD(timeout);
             ^
[... 470 lines cut ...]
                   ^
HAL.h:145:28: error: 'g_APinDescription' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                            ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
HAL.h:145:58: error: 'PIO_INPUT' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                                                          ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
HAL.h:145:97: error: 'PIO_Get' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                                                                                                 ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`

luc-github, 2016-09-08:
You use uno board and not Due
>`Arduino: 1.5.8 (Windows 8), Board: "Arduino Uno"

hletchworth, 2016-09-09:
Thank you. Right after I posted I double checked my board, as you suggested
I needed to change it.

Thanks!

closed 2016-09-09 by luc-github