luc-github/Repetier-Firmware-4-Davinci #188: 0.92-10rc1 Movement Command Strangeness

tdelany, 2016-09-08:
I have a DaVinci 1.0A that I have updated to the 0.92-10rc1 firmware.  I have noticed that I am having problems with the G1 GCode after certain other GCodes are sent to the printer.

One that definitely causes the issue is the M100 (clean head) GCode.  I use Simplify 3D with my printer and I have noticed that after sending the M100 GCode, the jog controls on the interface no longer work.

When, for example, you press the Z +10 button in the Simplify 3D interface, it sends a G1 code to the printer in the following format:  G1 Z10.000 F1200

I have discovered by trial and error, that if I manually send a G1 code to the printer that specifies all 3 axes, (e.g. G1 X0 Y0 Z0), the firmware will again respond normally to the short form of the G1 code sent by Simplify 3D.

Odd that.

Tom

luc-github, 2016-09-08:
Hi I just connected my printer to Simplify 3D and send M100 using S3D
wait for the process to be done - then hit JOG command and no issue

Can you give all information about your printer settings and what are your S3D settings 
mine :
[image]

and a detailed step by step way to reproduce issue like :
1 - power printer
2 - open S3D
3 - go to machine control panel
4 - connect to printer 
5 - etc..

tdelany, 2016-09-08:
ok.  Here are my settings:

[image]
[image]
[image]
[image]
[image]
[image]

Shortest steps to reproduce:

1. Turn on printer.
2. Start Simplify 3D
3. Open "Machine Control Panel"
4. Connect to printer (if not already - mine connects automatically).
5. Click on "Jog Controls" tab.
6. Click on "+Z 10" button.  Print bed moves down normally.
7. Click on "Communication" tab.
8. Enter "M100" at bottom and click on "Send" button. (see screenshot below). Wait for printer to go through nozzle cleaning gyrations.
9.  Click on "Jog Controls" tab.
10. Click on "+Z 10" button.  Nothing happens with the print bed.

From "Communication" window:

SENT: **M100**
READ: ok 0
READ: SelectExtruder:0
READ: FlowMultiply:100
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
SENT: M105
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: X:-33.00 Y:-12.00 Z:-0.000 E:0.0000
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: busy:processing
READ: ok 0
SENT: M105
READ: T:42.88 / 0 B:59.00 / 0 B@:0 @:0
...

SENT: G91
SENT: **G1 Z10.000 F1200**
SENT: G90
READ: ok 0
READ: ok 0
READ: x3
READ: X:-33.00 Y:-12.00 Z:15.000 E:0.0000
READ: ok 0
SENT: M105
READ: ok 0
READ: T:42.66 / 0 B:58.50 / 0 B@:0 @:0
READ: wait
SENT: M105
READ: ok 0
READ: T:42.66 / 0 B:58.44 / 0 B@:0 @:0
READ: wait

What else do you need?

tdelany, 2016-09-08:
I've discovered that this sequence triggers the issue as well:

1. Turn on printer.
2. Start Simplify 3D
3. Open "Machine Control Panel"
4. Connect to printer (if not already - mine connects automatically).
5. Click on "Jog Controls" tab.
6. Click on "+Z 10" button. Print bed moves down normally.
7. Click "Home All" button.
8. Click on "+Z 10" button. Nothing happens with the print bed.

"Home All" appears to send:
G28 X0 Y0 Z0

After the printer becomes unresponsive, manually sending something like "G1 X0 Y0 Z10" from the "Communication" tab will cause it to respond to buttons on the "Jog Control" tab again.

Tom

P.S.  Sometimes, seemingly randomly (I haven't been able to pin down a pattern), the printer will begin responding again on its own after some unknown amount of time.  Again, this seems to be pretty random.

tdelany, 2016-09-08:
Sorry.  I hit the wrong button.  Closed by accident.

luc-github, 2016-09-08:
I am able to reproduce after a power cycle only
pushing +z10 do nothing but pushing again and no issue

I was able to reproduce on S3D and Repetier

So seems the command is "lost" somewhere -  will dig in

tdelany, 2016-09-09:
Ok Luc.  If there is anything I can do to help, please let me know.  I can pretty much duplicate the issue with my printer at the drop of a hat.

It is interesting that you mention that it works in a macro.  I have been printing with the firmware without issue.

What got me started on this path was that after I upgraded the firmware, I was printing in air when I tried to start a print.  I noticed that after the nozzle-clean (M100) in my startup script, the bed was not moving back up like it had in the past, and it would start printing with the bed way down where the M100 had left it.

I played around and discovered what I reported to you.  I added a G1 movement code to work-around the issue.

Here is my startup script now:

G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 ;Home extruder
G1 Z15.0 ;move the platform down 15 mm
G92 E0 ;zero the extruded length
M100 ;Clean the nozzle
G4 P2
G1 X0 Y0 Z2.880 ;Added this to work-around movement weirdness in 0.92-10rc1

Tom

tdelany, 2016-09-09:
P.S.  The Z of 2.88 is because I print with a Z offset in S3D of 2.88 mm because I clip a piece of borosilicate glass to the top of my bed for printing.  It makes a nice level print surface (the original DaVinci bed is not flat - it is slightly warped).

Tom

luc-github, 2016-09-09:
Thanks - will check tonight or tomorrow as I am out all day

luc-github, 2016-09-09:
if you have time can you check if M100 T0 has same issue?

I will check tonight anyway

tdelany, 2016-09-09:
Luc,

Yes.  "M100 T0" behaves exactly the same way.  Once I send it, the jog controls no longer do anything until I send something like "G1 X0 Y0 Z10" (I have used different x, y and z values, and they all cause it to recover).

Tom

luc-github, 2016-09-10:
Hi I have add a missing command that seems to be the root cause with M100 - at least I cannot reproduce anymore
for the second method you provided - I am not able to reproduce the issue

Can you check ? Here the change : https://github.com/luc-github/Repetier-Firmware-0.92/commit/0d56084a48a941b7ba40c01ce6a4170f332a28c9

tdelany, 2016-09-10:
Luc,

I got tired of downloading zip files, so I finally downloaded git and cloned the repository.

I flashed my printer from the master, and I am sorry to report that I see no difference.

1. Power on printer.
2. Load Simplify 3D
3. Connect/verify connection to printer.
4. Press "+Z 10" jog button.  All works as expected.
5. Send "M100" from "Communication" tab.
6. Once the printer gyrations stop, press "+Z 10" jog control.  Nothing appears to happen.
7. Send "G1 X-33 Y-12 Z150 from "Communication" tab.
8.  Printer Z axis moves to 150 position.
9. Jog controls again function normally.

I reloaded the firmware twice to make certain that I was loading the correct code.  I even opened printer.cpp and verified that the line that you added was present in the file at line 313.

What else can I try for you, mon ami?

Tom

luc-github, 2016-09-10:
Well I do not know as I cannot reproduce anymore:(
I saw your Z acceleration is high
>G1 Z10.000 F1200

can you decrease speed to 2.0 to have a F120 ?

luc-github, 2016-09-10:
I will flash my 2.0 as a 1.0 to see if it change anything

luc-github, 2016-09-10:
I cannot reproduce - tried many time - the only difference I have with you is the baudrate - mine is 230400 on printer and 230400 on S3D

tdelany, 2016-09-10:
Re: your comment:
>Well I do not know as I cannot reproduce anymore:(
>I saw your Z acceleration is high
>
>G1 Z10.000 F1200
>can you decrease speed to 2.0 to have a F120 ?

That gcode sequence is generated by S3D when I press the Jog button.  Can you point me to a setting to reduce that?

>I cannot reproduce - tried many time - the only difference I have with you is the baudrate - mine is 230400 on printer and 230400 on S3D
 
I will try reducing my baudrate to match yours.

Tom

luc-github, 2016-09-10:
here :
[image]

tdelany, 2016-09-10:
Well I'll be damned.  I switched the baudrate to 230400 and it stopped failing!

luc-github, 2016-09-10:
Ok so I was not so wrong ^_^

luc-github, 2016-09-10:
but this is weird if it was ok before upgrading

luc-github, 2016-09-10:
I will recheck the serial setting process as virtual EEPROM has been rewritten for latest may be only 230400 the default is used and others changed are not taken in acccount

tdelany, 2016-09-10:
Interestingly, it still won't respond to the jog controls after I click on "Home All", but sending M100 now restores its function!  Now after sending M100 everything works again.  Bizarre, but progress.

Interestingly, this only seemed to affect printing in that it messed up the way my starting and ending scripts worked.  Otherwise printing was fine (although I seemed to get more pauses than before.  That might also have something to do with the baudrate.

Yes, 250000 worked with the old firmware version "Repetier-Firmware-0.92-0.92.3.0".

I'll keep printing and let you know how things go.  A bientôt!

Tom

luc-github, 2016-09-10:
Yes at least some progress 
I will check the code difference for home  and serial setting - at least it is a clue

tdelany, 2016-09-10:
So far so good as the printing goes. 

I don't envy you.  You not only have your own code modifications to worry about - you are also at the mercy of the Repetier folks and all of the things that they might change/break.

Tom

luc-github, 2016-09-11:
Ok after reviewing reviewing the code and remember my todo list Serial speed issue was a potential issue that I forget to confirm and fix I already raised the point to repetier (https://github.com/repetier/Repetier-Firmware/commit/b3286c056ed9580809a60a04755b007adfc878f1#commitcomment-18245778) then forget it as I never use USB connection

My bad - will fix it tomorrow

luc-github, 2016-09-11:
For better tracking I have created a separate issue : https://github.com/luc-github/Repetier-Firmware-0.92/issues/189

luc-github, 2016-09-12:
using 250000 as baud rate now it works : https://github.com/luc-github/Repetier-Firmware-0.92/commit/b64b0701c86de23144d0eced2db32fb3cb5161d5

Still cannot reproduce the other way :disappointed:

tdelany, 2016-09-12:
I printed quite a bit over the weekend at the lower baudrate without issue.  I will try the faster again with the new build when I get home.

I cannot explain the oddness that I have seen that you can't replicate.  I have learned to work around it, and it doesn't seem to affect normal printing.  When I have time, if you like, I can try various combinations of things to try to nail down what I am seeing.  At this point, I would not consider it a high priority for you, though.

Tom

luc-github, 2016-09-12:
Yes - anyway take your time - this week end I go to vacation - I will be 10K km from home and printer so no chance to reproduce it even you give me the steps ^_^, will be back beginning of October :smile:

luc-github, 2016-10-06:
Any update on this ?

tdelany, 2016-10-08:
> 
> Any update on this ?
Sorry.  Been preoccupied with the Hurricane and all.

Will let you know as soon as I can.

Tom

tdelany, 2016-10-23:
Sorry it took so long to reply to this.  The communication seems to now be working at 250000 at least as well as it did at 230400.

I have printed may things with the updated firmware without issues.  Although I still have some weirdness with the jog controls in Simplify3D, the actual printing is fine.  I have learned how to work around the jog control strangeness when I encounter it (manually sending something like:  "G1 X-33 Y-12 Z10" seems to "reset" things.

luc-github, 2016-10-23:
at least some progress, what are the `weirdness ` with Jog ?

luc-github, 2016-12-29:
if no update I close issue

closed 2016-12-29 by luc-github

tdelany, 2016-12-29:
Thanks Luc.  Sorry.  I've been busy with other stuff.  It's working well enough and I have been printing with it without problems (does not seem to affect the actual printing).  I documented the jog weirdness before, and you could never reproduce it.  I agree with closing the issue.

luc-github, 2016-12-29:
Thank you - Happy new year :fireworks: