luc-github/Repetier-Firmware-4-Davinci #189: EEPROM serial speed is not used - only 230400 is active (default one)

luc-github, 2016-09-11:
because of new virtual EEPROM -  Serial is set befaore EEPROM is loaded so default value is used

tdelany, 2016-09-11:
Thank you, Luc.

luc-github, 2016-09-12:
should be fixed by https://github.com/luc-github/Repetier-Firmware-0.92/commit/b64b0701c86de23144d0eced2db32fb3cb5161d5

closed 2016-09-12 by luc-github

tdelany, 2016-09-12:
I'll check it out when I get home later.  I am back to work this week. :-(

luc-github, 2016-09-12:
No problem - take your time - thanks

TGMods, 2016-09-12:
Hi guys,
This morning I downloaded the latest master. I spent over an hour reflashing my Duo at various baud rates and was able to connect at whichever baud rate I had selected. I am now running my Duo at 250000.
Hope this helps

luc-github, 2016-09-12:
Thanks a lot for your feed  back :smile: