luc-github/Repetier-Firmware-4-Davinci #190: please help so close but dont understand

naothan, 2016-09-13:
In file included from Repetier.h:572:0,
                 from BedLeveling.cpp:107:
HAL.h: In static member function 'static void HAL::serialSetBaudrate(long int)':
HAL.h:684:14: error: 'class UARTClass' has no member named 'setInterruptPriority'
       Serial.setInterruptPriority(1);
              ^
Error compiling

luc-github, 2016-09-13:
Looks like you are using IDE 1.5.8 to compile but as mentioned in readme:
`Use an arduino IDE supporting arduino DUE, 1.6.9 with Due 1.6.8 module from board manager.`

luc-github, 2016-09-13:
latest version of repetier need 1.6+ IDE

naothan, 2016-09-13:
have tried them all when i use a newer version the board isnt there (native usb port)

luc-github, 2016-09-13:
yes you did not compile for Due but for Yun 
please check FAQ carefully : https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+

And you may review also this : https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

ironmann90, 2016-09-13:
I had the same issue earlier and it was because i was using 1.5.8 IDE. So i followed your steps on using a 1.6+ IDE. Now after flashing and loading everything (using 1.6.8DUE. I have 2 black bars and a msg pop up on my screen saying "USB Device not recognized"

luc-github, 2016-09-13:
well different issue which means variant.cpp is not correct,
but same answer:
please check FAQ carefully : https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+

And you may review also this : https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

luc-github, 2016-09-19:
any update on this?

luc-github, 2016-10-02:
if no update I close issue

closed 2016-10-02 by luc-github