luc-github/Repetier-Firmware-4-Davinci #192: Cleaning the print nozzle after powering on doesn't home the axes

cdoc83, 2016-09-14:
All of my prints leave the X and Y axes in the home position, and the Z axes at the max position (200).  If, after a print, I power down the printer, and sometime later power up the printer and immediately attempt to clean the nozzle, the Z axis attempts to lower, causing a terrible noise, and probably is on the verge of causing damage to the printer.

I believe the firmware should keep track of whether or not it has been homed since power up.  If it has not, it should at a minimum be required to home all axes.  It would be better if the firmware would just home the axes as needed.

luc-github, 2016-09-14:
well this is the case if not homed we home first: 
https://github.com/luc-github/Repetier-Firmware-0.92/blob/b64b0701c86de23144d0eced2db32fb3cb5161d5/src/ArduinoDUE/Repetier/Printer.cpp#L217-L222

Can you give me a detailed reproduction steps as I just checked and cannot reproduce it, after a power up I send a M100 command and it home first

cdoc83, 2016-09-14:
Ah, sweet, I see that's included in v0.92.10rc1, but wasn't in v0.92.3.0.  There is no such check that I can see in v0.92.3.0.

luc-github, 2016-09-14:
it is weird the check is present in 0.92.3 as well
https://github.com/luc-github/Repetier-Firmware-0.92/blob/v0.92.3.0/src/ArduinoDUE/Repetier/Printer.cpp#L184-L188

cdoc83, 2016-09-14:
Ahh, I was looking at printer.h.  My bad.  Still odd it's not working, maybe I'll throw some logging in there to see if it thinks it's homed but isn't.

luc-github, 2016-09-14:
0.92.10 is big rewrite, it may fixed it

cdoc83, 2016-09-14:
Don't know what the deal was...  I added a few logging lines in Printer.cpp, reflashed, and the problem is gone.  Oh well, thanks for your super quick responses!

closed 2016-09-14 by cdoc83