luc-github/Repetier-Firmware-4-Davinci #193: 1.0A Second Fan Clarification

DaAwesomeP, 2016-09-14:
Hi,
I saw the `REPURPOSE_FAN_TO_COOL_EXTRUSIONS` option in the firmware and explored some issues here. I have a big box fan that I've been using to cool off the extruder manually. If I understand correctly, there is an extra pin on the board that can be used to control this.

 - Is this pin simply high/low 5V? Can I use a relay (or PowerSwitchTail) to turn on and off the mains-voltage box fan?
 - Do I need to set `REPURPOSE_FAN_TO_COOL_EXTRUSIONS` to `1` to use the second pin or is that only to use the original fan to cool the extruder?
 - In the first picture of https://github.com/luc-github/Repetier-Firmware-0.92/issues/152#issue-149579186 there is a connector with a wire annotated as fan 2, but to the left is an entire connector I could solder to labeled "Fan2." Which pins/wires will be triggered)?

It'd be awesome if you could add a wiki page or some docs somewhere about this. I didn't see this feature of the firmware or 1.0A at first and it's definitely a plus! Thanks for packaging this awesome firmware!

luc-github, 2016-09-14:
Hi please check also https://github.com/luc-github/Repetier-Firmware-0.92/issues/174
actually have several way as you can see
I may not really being able to write as I have already the support of 2 fans as I have a 2.0
feel free to create a complete doc and share

DaAwesomeP, 2016-09-14:
I'd be happy to make a doc, but please answer my questions if you can. What config options specifically do I need to enable/specify, and how do I find the Due pin for a particular random pin on the machine?

Thanks for your swift reply!

DaAwesomeP, 2016-09-14:
OK, so I found `ORIG_FAN2_PIN` and `EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN`. Which of those need to be set and do I need to set `REPURPOSE_FAN_TO_COOL_EXTRUSIONS`?

luc-github, 2016-09-14:
REPURPOSE_FAN_TO_COOL_EXTRUSIONS is just a shortcut to define the others settings automaticaly
the 3 settings fan pin, extruder cooler and fan control
for the pins the reference are in issue you mentioned

closed 2016-09-22 by DaAwesomeP