luc-github/Repetier-Firmware-4-Davinci #194: Repetier reflash will not complete (Hangs)

bgauweiler, 2016-09-18:
I'm trying to upgrade my new Da Vinci 1.0A printer to your latest version of Repetier and I have not been able to get the upload in Arduino 1.6.9 (with 1.6.8 DUE board added) to successfully complete after several attempts from different Windows 7 computers.  Each time I try, the upload completes and then the "Verify" step will hang at either 79 or 88%.

Please let me know what other information I can provide to help troubleshoot my installation problem.

Bernie

luc-github, 2016-09-19:
Hi did you check FAQ (https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+) ?

especially : https://github.com/luc-github/Repetier-Firmware-0.92/issues/185

bgauweiler, 2016-09-19:
Thank you for pointing me to the FAQ.  I read #185 and disabled the verification in the Arduino preferences and in installation process completed, and I now have the Repetier FW loaded.

What additional steps would you recommend I take at this point to validate my installation?  I would like to understand the root cause of the problem I encountered with the verification.  Are there any troubleshooting steps I should take?

Bernie

luc-github, 2016-09-19:
I never experienced such issue - so it is just a guess: unexpected behavior may show corrupted upload, like random reset or anything not expected - really  hard to say.

about test, only valid verify is the one - so as suggested in FAQ: change usb cable and be sure no loose connector, also be sure cable to not cross any electric one or other electric device, etc...
Another thing could be a new wipe of FW to be sure it is properly wiped but again it is just a guess

bgauweiler, 2016-09-19:
I decided to reset the Flash memory again and exchanged the USB cable.  I
also used Arduino running natively on my Mac vs. running the PC version
under Windows 7 on my Parallels Desktop VM.  Your code compiled and loaded
perfectly.

Thanks for you help, and for your dedication to the Da Vinci community.

Bernie

luc-github, 2016-09-20:
Thanks I will add this to the FAQ as it seems not the first time I read people has issue flashing using VM

bgauweiler, 2016-09-20:
I actually ran into the same issue when I was trying to compile and verify
from my PC laptop.  The significant difference is that I replaced the USB
cable this time.  I believe that fixed the issue.

Bernie

closed 2016-09-21 by bgauweiler