luc-github/Repetier-Firmware-4-Davinci #195: Davinci 2.0 Duo, cant quite get it to work.

Seiler270, 2016-09-20:
Im trying to flash via the directions here: 

https://github.com/luc-github/Repetier-Firmware-0.92

Im a bit confused by step 4:

""""Update variants.cpp arduino file with the one present in src\ArduinoDUE\AdditionalArduinoFile according your IDE version""""

Ok so i can find the variant.cpp file in the repetier folder  Lacated

C:xxxxxxxxxxxxx\Repetier-Firmware-0.92-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.6.9 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

The issue is, where to I place it in the Ardino program group?

C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants  does not contain a variant.cpp files..

so where does it go?

I tried skipping the step, and continuing on and everything goes along as it should but after the firmware upload and the reboot i get nothing,  its not even recognized by windows after that.  I have to jumper J37 and start over.

Thanks for any help!

luc-github, 2016-09-20:
replacing variant.cpp is mandatory, please check FAQ : https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+
and this https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

Seiler270, 2016-09-20:
Thanks for the fast reply!

I understand that replacing the variant.cpp is important..  

but WHERE does it go..   

I read the link you posted and it is not quite the same issue, its backward.

I know where the variant.cpp file is in the repetier folder...

It says to REPLACE the file,  where is the one that needs to be replaced?  there is NOT a variant.cpp file under C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants..    which is where I expected to find it..

luc-github, 2016-09-20:
that is why I sent you links, variant.cpp is not in program files but users files - please read carefully BOTH links content

Seiler270, 2016-09-20:
I read it

Stupid windows was hiding the DIR.

I got everything moved, flashing now.

It flashed fully and is now verifying, but its hanging out on 89%,  ill give it a bit longer but I think I remember reading something about this.

luc-github, 2016-09-20:
Yes it is FAQ too

Seiler270, 2016-09-20:
Well it just stayed at 89% for several minutes so I closed arduino and shut the printer off.

powered it back on and there are no longer any black bars, just a blank screen.

I tried to reflash again with verify disable and it went through and said CPU reboot, but still nothing..

nothing whatsoever.

Seiler270, 2016-09-20:
ok so now i cant jumper it back to black bars... 

All i get is a completely blank screen, nothing else.

Seiler270, 2016-09-20:
just ran it again with  verification on, it went through just fine.

still just a blank screen, no activity.

Seiler270, 2016-09-20:
Out of ideas.

Seiler270, 2016-09-20:
Well i started from scratch and did the whole process over again.

Verify successful
done in 27.543 seconds
Set boot flash true
CPU reset.

nothing, blank screen..  thats it...

so... wtf.

Seiler270, 2016-09-20:
the printer shows up under device manager as Arduino Due...  so...  its there  but still nothing

Seiler270, 2016-09-20:
this is odd..

Repetier host can see the printer, shows it as idle...  

but nothing works/does anything.

luc-github, 2016-09-20:
do you have a 2.0 or a 2.0A ? (2.0 = normal SD, 2.0A =micro SD)

Do you see EEPROM.bin from repetier SD card management? if yes delete it and power off printer, if no remove SD card and delete EEPROM.BIN from PC

Seiler270, 2016-09-20:
I have the 2.0A with the micro SD

Where am I looking for these files?
in the arduino program?

luc-github, 2016-09-20:
On SD card

Seiler270, 2016-09-20:
Ok deleted EEPROM.BIN from the printer's SD card, reinsalled card and turned on printer.

nothing..

reflash?

luc-github, 2016-09-20:
what means nothing  ?

Seiler270, 2016-09-20:
the printer does nothing, no beeps no light nothing.

luc-github, 2016-09-20:
yes seems flash is not done properly somehow

Seiler270, 2016-09-20:
ok ill do it again

Seiler270, 2016-09-20:
Reflashed.

Same issue

No image on screen.

When trying to print the bed heats up but the extruders never do and it kicks back the error saying 

fatal: heator/sensor failure printer stopped and heaters disabled due to this error.

I did notice one thing, 

when I go to print It says:  Select Extruder 0  instead of 1 or 2..   is that maybe part of the problem with the heating?

luc-github, 2016-09-20:
As you cannot see screen you cannot load fail-safe settings but did you send M502/M500 instead as writed in readme ?

Seiler270, 2016-09-20:
Yes, and when  I did the printer light came on.

luc-github, 2016-09-20:
and you should have sound when press home button right ?

Seiler270, 2016-09-20:
Yes it beeps

luc-github, 2016-09-20:
so better than no light no sound like before

what did you change in configuration.h for your printer ?

Seiler270, 2016-09-20:
Just the definitions at the beginning 

Define Davinci
Model.

I set Davinci to 3
Model to 2

Seiler270, 2016-09-20:
looking at the config..

there is the line 

Set to 1 if all extruders use the same heater block. Temp. control is then always
controlled by settings in extruder 0 definition. */

#define SHARED_EXTRUDER_HEATER 0

mine has one heater for both extruders im pretty sure,  so I would think this should be set to 1 not 0.

luc-github, 2016-09-20:
No, 2.0A does not have SHARED_EXTRUDER_HEATER, it has 2 extruders each one using one heater

All I can see until now, it is that you do not follow strictly what it is mentioned in readme / https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

I suggest you to remove all arduino files/ delete FW files
reinstall arduino 1.6.9 / due 1.6.8 and FW and slowly / carefully follow each step of readme or https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

As I cannot imagine others change you may have done to get the lcd not working, I really suggest you to stick to what is written as you seems to do others changes on your own that affect the fw which work on others 2.0A - this is the best advice I can give

Seiler270, 2016-09-20:
will start over again.

Seiler270, 2016-09-20:
Ok Im typing this as I do it, so I make sure I dont miss anything.

Turn off printer
Remove SD card from printer and delete EEPROM.BIN from card

Reinstall SD card and power printer on.

No pic on screen, no lights no sounds.

Install  arduino-1.6.9-windows
Install complete

Open arduino

Select arduino AVR boards version 1.6.12 as shown is your picture.
Install

Select arduino SAM boards 1.6.8 as shown in your picture.
Install

Close arduino

Open Folder:  Repetier-Firmware-0.92-master

Navigate to:  \src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.6.9 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

Copy variant.cpp

Navigate to: 
C:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

 Paste variant.cpp in dir, replace existing file.

Printer on, USB connected, correct Com port verified, good connection

Open arduino

Select Tools

Board

Select  arduino Due (Native USB port)

Select Port
COM6 (arduino native USB)

from arduino file menu open   \Repetier-Firmware-0.92-master\src\ArduinoDUE\Repetier
Repetier arduino file.

File loaded.

Select Configuration Tab

#define Davinci 3
#define model 1

No other changes made,

Select Verify

Message:  const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS; // Sensitive pin list for M42

Done compiling

Select UPLOAD

Upload complete, Verifying

Verify successful
done in 42.156 seconds
Set boot flash true
CPU reset.

Reboot printer

No image on screen at this point

Close arduino 

Install repetier Host

Install repetier Server

Launch repetier host.

Open server front end.

Add printer

Select correct com port,

Connection success,  continue.

Fetching configureation from firmware

Goes to next page automoatically

All values are 0  (????????)
Z-max is 0 and is in RED, cannot continue...

So this is where I am at, cannot progress further.

It seems the server is not pulling the printer info from the firmware..

Is everything I have done up to this point correct?

Seiler270, 2016-09-20:
Disconnect from printer

Start local server

open server front end

add printer

select correct com port

Connection success

retrieving info from firmware complete

Step 4

Number of extruders:  2
Filiment diameter 1.75

heated bed installed yes

step 5 features
Fan installed: no   SDCard reader: yes   Software power: no

FINISH

Select printer,  control from server

Test heated bed:  good  
Test Extruder 1:   extruder does not heat
Test Extruder 2:   extruder does not heat

Fatal Error:  heater sensor failure message

Screen still blank

Return to host

Connect to printer VIA  repetier server

load object to print

slice with slicer

select print 

Heating bed:

Bed heats to 70c

Heating Extruder 1

Log shows
12:46:30.373 : FIRMWARE_NAME:Repetier_0.92.10 FIRMWARE_URL:https://github.com/luc-github/Repetier-Firmware-0.92/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:DaVinci EXTRUDER_COUNT:2 REPETIER_PROTOCOL:3
12:46:30.373 : Printed filament:0.00m Printing time:0 days 0 hours 0 min
12:46:30.373 : PrinterMode:FFF
12:46:30.420 : Error:M114
12:46:30.420 : X:NAN Y:NAN Z:NAN E:NAN
12:46:30.420 : DebugLevel:6
12:46:30.420 : SelectExtruder:0
12:46:30.436 : Begin file list
12:46:30.483 : ad_check_0002 1400
12:46:30.498 : Machine_Life.dat 14
12:46:30.498 : SAMPLE01.gcode 661731
12:46:30.498 : SAMPLE02.gcode 396439
12:46:30.514 : SAMPLE03.gcode 3669461
12:46:30.561 : last_temperature.txt 4
12:46:30.577 : offline_print.gcode 1425949
12:46:30.577 : eeprom.bin 4096
12:46:30.577 : End file list
[... 121 lines cut ...]
13:01:05.440 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.440 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.440 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.671 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.671 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.672 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.672 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.672 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.672 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:05.783 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.002 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.
13:01:06.216 : fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.

So I m back to the same point,  nothing on the screen, and the extruders wont heat.

Seiler270, 2016-09-20:
I posted this issue with the heating on the repetier forums and it was suggested that the correct heater output pin may not be selected.

But Im not sure where to look for that.

luc-github, 2016-09-20:
I do not think it is the issue - it is working on other 2.0A see https://github.com/luc-github/Repetier-Firmware-0.92/issues/94 you can use same https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.3.0 to verify the 0.92.10 is not broken for 2.0A which I doubt as it is working on 1.0A/AiO  

About heating I guess more about hardware failure check https://forum.voltivo.com/showthread.php?tid=8517

About screen, nothing = no backlight + no char or backlight + no char ?

Seiler270, 2016-09-21:
trying the version you posted.

I highly doubt anything is wrong with the hardware considering everything was working perfectly just before flashing this firmware.

luc-github, 2016-09-21:
what about screen ?

well short the davinci is easy - just use metal brush too strong cleaning noozle and it is done, another possiblity is connector loose - but this should be detected if disable decouple test allow to heat

Seiler270, 2016-09-21:
setting FEATURE_DECOUPLE_TEST   to 0 causes a failure to compile.

Extruder.cpp:409: error: 'isSensorDecoupled' was not declared in this scope

  if(isSensorDecoupled())

luc-github, 2016-09-21:
on which version of FW ?

Seiler270, 2016-09-21:
https://github.com/luc-github/Repetier-Firmware-0.92

luc-github, 2016-09-21:
I guess you mean master branch based on 0.92.10 and not the link I gave you based on 0.92.3

Seiler270, 2016-09-21:
ill try that one.

Seiler270, 2016-09-21:
trying now with 
the firmware you linked.

bed heats

then it shows 
21:48:01.943 : Unknown command:N85 M107
then says heating extruder and just stays there, extruder never heats up.

luc-github, 2016-09-21:
M107 unknow command is normal - Davinci does not allow fan control

can you share your begining part of GCODE code your are using for printing ? the part which set temperatures for extruders

what about screen ?

luc-github, 2016-09-21:
Repetier FW and Host use T0 and extruder 0 for first extruder and T1 and extruder 1 for second one

Seiler270, 2016-09-21:
beginning part of GCODE when printing:

22:08:04.466 : OpenGL version:4.5.0 NVIDIA 372.54
22:08:04.466 : OpenGL extensions:GL_AMD_multi_draw_indirect GL_AMD_seamless_cubemap_per_texture GL_AMD_vertex_shader_viewport_index GL_AMD_vertex_shader_layer GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compatibility GL_ARB_compressed_texture_pixel_storage GL_ARB_conservative_depth GL_ARB_compute_shader GL_ARB_compute_variable_group_size GL_ARB_conditional_render_inverted GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_indirect GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_ES2_compatibility GL_ARB_ES3_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_2_compatibility GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_fragment_shader_interlock GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_gpu_shader_int64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_post_depth_coverage GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_robustness GL_ARB_sample_locations GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_clock GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_sparse_texture2 GL_ARB_sparse_texture_clamp GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_cube_map_array GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_filter_minmax GL_ARB_texture_float GL_ARB_texture_gather GL_ARB_texture_mirror_clamp_to_edge GL_ARB_texture_mirrored_repeat GL_ARB_texture_multisample GL_ARB_texture_non_power_of_two GL_ARB_texture_query_levels GL_ARB_texture_query_lod GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_texture_rgb10_a2ui GL_ARB_texture_stencil8 GL_ARB_texture_storage GL_ARB_texture_storage_multisample GL_ARB_texture_swizzle GL_ARB_texture_view GL_ARB_timer_query GL_ARB_transform_feedback2 GL_ARB_transform_feedback3 GL_ARB_transform_feedback_instanced GL_ARB_transform_feedback_overflow_query GL_ARB_transpose_matrix GL_ARB_uniform_buffer_object GL_ARB_vertex_array_bgra GL_ARB_vertex_array_object GL_ARB_vertex_attrib_64bit GL_ARB_vertex_attrib_binding GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_vertex_type_10f_11f_11f_rev GL_ARB_vertex_type_2_10_10_10_rev GL_ARB_viewport_array GL_ARB_window_pos GL_ATI_draw_buffers GL_ATI_texture_float GL_ATI_texture_mirror_once GL_S3_s3tc GL_EXT_texture_env_add GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_compiled_vertex_array GL_EXT_Cg_shader GL_EXT_depth_bounds_test GL_EXT_direct_state_access GL_EXT_draw_buffers2 GL_EXT_draw_instanced GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXTX_framebuffer_mixed_formats GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_packed_pixels GL_EXT_pixel_buffer_object GL_EXT_point_parameters GL_EXT_polygon_offset_clamp GL_EXT_post_depth_coverage GL_EXT_provoking_vertex GL_EXT_raster_multisample GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separat [... cut ...]
22:08:46.403 : <Slic3r> => Preparing infill
22:08:46.403 : <Slic3r> => Infilling layers
22:08:46.622 : <Slic3r> => Generating skirt
22:08:46.622 : <Slic3r> => Exporting G-code to composition.gcode
22:08:46.731 : <Slic3r> Done. Process took 0 minutes and 0.552 seconds
22:08:46.731 : <Slic3r> Filament required: 144.9mm (0.3cm3)
22:08:52.566 : Unknown command:N25 M107
22:09:49.343 : X:0.00 Y:0.00 Z:0.000 E:0.0000

Thats all its showing now because i still have decoupling disabled, shall I reinable it?

luc-github, 2016-09-21:
not this is not your GCODE - this is what repetier display
you must export the GCODE file instead of send to printer

luc-github, 2016-09-21:
https://www.repetier.com/documentation/repetier-host/gcode-editor/

Seiler270, 2016-09-21:
ok let me try that.

also just loaded the .92 master with decoupling enabled again.

Seiler270, 2016-09-21:
is this what you meant?

generated by Slic3r 1.2.9 on 2016-09-20 at 22:18:31

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.48mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.48mm
; top infill extrusion width = 0.48mm

M107
M190 S70 ; set bed temperature
M104 S211 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

M109 S211 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.350 F7800.000

luc-github, 2016-09-21:
Yes - ok I do not see anything wrong

using repetier to heat extruder does not work neither, right ?

again what about lcd screen ? no backlight + no char or backlight + no char ?

are the 3 red led working on your board ?

Seiler270, 2016-09-21:
nothing on the screen at all, no backlight nothing.

using repetier the extruders will not heat, but the bed will

there are 2 red led on board that are on.

luc-github, 2016-09-21:
ha finally 
2 and not 3 = fuse issue
 no LCD and no extruder are the consequences

Seiler270, 2016-09-21:
well thats GREAT news... so...  where is the fuse?  hahahahaha

luc-github, 2016-09-21:
read this https://forum.voltivo.com/showthread.php?tid=8498&highlight=lcd

and the previous one I gave you : https://forum.voltivo.com/showthread.php?tid=8517&highlight=extruder

luc-github, 2016-09-21:
for hardware issue I suggest to post to forum voltivo or soliforum to get help as I am not electronician and not best person to help on this

Seiler270, 2016-09-21:
looks like I just need to solder j4 shut, looked up a video of it, seems simple enough.

Ill do that real fast and we will see what we have.

luc-github, 2016-09-21:
Ok good luck

Seiler270, 2016-09-21:
Looks like I need a new motherboard. 

Im not getting 12v reaching JP4,  so its not a issue of that resistor being blown, its a matter of it not getting voltage at all in the first place.

i guess Ill see what xyz says.

luc-github, 2016-09-21:
sorry for you - Davinci is very fragile and easy to get shorts

luc-github, 2016-10-02:
if no update I close issue

closed 2016-10-02 by luc-github