luc-github/Repetier-Firmware-4-Davinci #197: Not completing. Brick?

higuerajoey, 2016-09-29:
I went through the install process on my 1.0 3F10a and at the end after the arduino code compiled and uploaded, the printer kept the 2 black bars and wouldn't connect to the computer for longer than a couple seconds. Did the J37 erase and tried again. same issue.

luc-github, 2016-09-29:
DId you checked the FAQ ?

higuerajoey, 2016-09-29:
Sorry about that, just did it, now getting this error, 
[image]

higuerajoey, 2016-09-29:
@luc-github sorr for pestering, I'm just freaking out because i was trying to do this to the printer in the university lab I work in, and I really don't want to have ruined a  university printer

luc-github, 2016-09-29:
this happen sometimes with IDE when you install/remove board but acction is not done properly
just exit IDE and reopen check in board manager the board is really present

luc-github, 2016-09-29:
You cannot bricked the printer by just uploading the FW - Davinci is very fragile from HW point of view so be nice to it, but even HW issue are common, this is not what happen here

higuerajoey, 2016-09-29:
@luc-github Thanks for being so helpful, you're the best. 

ok, so I replaced the files that were mentioned in the FAQ and re-installed the board from the board manager and it uplloaded to the printer this time, but the same thing is happening where it won't stay connected to the computer

luc-github, 2016-09-29:
and black bars ?

higuerajoey, 2016-09-29:
Yes, black bars

luc-github, 2016-09-29:
files replacement must be done after any update as everything is erased

Black bar just means variant.cpp is not the correct location or properly set like described in FAQ

luc-github, 2016-09-29:
Additionnaly do not forget to power off / power on the printer after flashing

higuerajoey, 2016-09-29:
I must not be installing the Variant.cpp in the correct place because it is doing the same thing, even after turning it off and on. here is the location I'm installing it 

C:\Program Files (x86)\Arduino\hardware\sam\1.6.8\variants\arduino_due_x

luc-github, 2016-09-29:
Please take time to read the FAQ everything is there - yes location is wrong

luc-github, 2016-09-29:
or read https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

higuerajoey, 2016-09-30:
Awesome, Finally got it. I had to redownload the cpp files and then reinstall them in the proper folder. 

Thanks for putting up with me.

luc-github, 2016-09-30:
good happy printing 😄

closed 2016-09-30 by luc-github