luc-github/Repetier-Firmware-4-Davinci #198: Will Repetier work on Da Vinci 1.0A with firmware 1.0.6?

stormbeer1, 2016-10-02:
I got a new Da Vinci 1.0A with firmware 1.0.6 this version will not let me set the extruder temp after I flash the cartridge eeprom, the reset works for a new 240M spool etc but the the stock 210C temp remains no matter what I do in the arduino reset sketch I.E. cannot set temp to 230C etc. I got a new spool of aftermarket filament but it needs around 230C so it keeps "clicking" my extruder because it cannot heat it up enough. I have heard of the slic3r software and their workarounds editing the G-Code in Notepad++ but this seems like a hassle. Is there another software alternative to XYZware that will let me set the extruder temp without risking repetier bricking? Has anyone flashed my particular model with FW1.0.6 with Repetier successfully?? If so please drop me a line and tell me your story and how you did it. I refuse to pay for XYZ expensive 600g refills which are just a little over a half of a typical 1Kg spool that cost only about $20 versus $30+ for an XYZ 600g spool thats just ridiculous, please message me, Help!

luc-github, 2016-10-02:
Once Firmware is wiped by jumper - there is nothing left from FW point of view - what ever the FW version is.
Any version of XYZ fw can be wiped by jumper because it is done a HW level and not FW level.

So yes will works.

stormbeer1, 2016-10-02:
Thanks for the response Luc. My concern is not wiping it but that the motherboard itself might be rigged like the firmware,.. so you cannot flash repetier and just brick it, I have heard rumors. Do you have one of these 1.0A with fw 1.0.6? Or do you know anyone who has flashed one with repetier? let me know buddy thx!

luc-github, 2016-10-02:
The only issue is not following the steps properly - always user issue 
the board is not locked : https://forum.voltivo.com/showthread.php?tid=7941&pid=7456#pid7456

stormbeer1, 2016-10-02:
Thanks luc!! can you please tell me if I should use Arduino 158 with Repetier .92.3  with the 158 compatible hardware files etc
or go with the Arduino 169 with the new .92.10 Repetier? let me know buddy thanks again!!

luc-github, 2016-10-02:
Up to you 

0.92.3 is now old (1 year at least) but still stable if you do not use Autolevel and  do not use new features

0.92.10 is the active one - has more features  and have better autolevel

stormbeer1, 2016-10-02:
Ok will do my friend.  I am confused about how to update the variant.cpp files etc and what files to copy into the Arduino install folder but auto level sounds like it is worth it. So I will get cracking and do my homework. If you have any more advice or tips let me know and thanks again for letting me know what I needed to know so fast THANK YOU!

luc-github, 2016-10-02:
I suggest you to read carrefuly the FAQ before doing anything

stormbeer1, 2016-10-02:
OK THANKS AGAIN!

stormbeer1, 2016-10-03:
Flashed with 158 so happy right now, thanks again my friend. Now I have to learn how to use the Repetier host program and adjust the bed settings etc so my bed will not collide with the nozzle, Luc you the man!!

jpalovick, 2016-10-03:
Yes it works.  I just did the same thing.  My only comment would be to say that variant.cpp and usbcore.cpp are in the Appdata section.  The easiest way to do that is to open the File->Preferences and near the bottom there is a link to preferences.txt.  If you click that link it will take you to the directory structure that needs updating.  I changed the names of the original files and copied the new ones into their respective directories.

luc-github, 2016-10-06:
If no update I close issue

closed 2016-10-06 by luc-github

stormbeer1, 2016-10-06:
Update I just flashed .92.10 it was easy now I need help  bed settings so it will not hit the extruder.  The old .92.3 videos  are different than .92.10 where can I find the settings for my Da Vinci 1.0A with Repetier .92.10 and repetier host 1.6.2? thanks guys!