luc-github/Repetier-Firmware-4-Davinci #200: Lightning strike

Monster262, 2016-10-05:
I know I am leaving this in the wrong area but perhaps someone can help reroute me. Had a terrible storm in our area which took out mtg router and modem.thought it was the only casualty but fired up my 2.0A and I get black lines and nothing is recognizable. No USB, no sound, no lights.. Black lines on display. Any suggestions?

luc-github, 2016-10-05:
I would say check the 3 leds on board are on, check mcu surface to see if damaged or not.
Use jumper to wipe FW which seems damaged, this should allow you to be detected by PC and flash again

Monster262, 2016-10-05:
Leds are on. MCU appears undamaged. Jumpered J37 and began process of reinstalling repetier. Still no signs of life. I'll keep you posted.

luc-github, 2016-10-19:
If no update I close issue

closed 2016-10-19 by luc-github