luc-github/Repetier-Firmware-4-Davinci #202: Davinci 1.0 no longer extruding filiment

Chromewrecker, 2016-10-11:
Ok so not really sure what happen but everything was fine until last night. I was in the middle of a print and messing with the computer when a video card error popped up and the computer froze. After a minute of the screen flashing on and off the repetier host was not responding and the print stopped. Was I reset the computer and the printer I retried printing the file it started off fine so I went to bed woke up this morning with a bout one inch of the file printed a inch of mess and then nothing above that and the extruded was still making its passes about a four inches above that. So I removed the extruded gave it a good cleaning and reset the bed started the print again this time it would print but it was skipping. So I reset it and it got worse. Every time I kill the print and try it the issue got to the point of nothing coming out. So I cleaned it again tried different filiment started the prin. The first layer printed then started skipping again then nothing. When I lightly place my fingers on the filiment during a print i can feel it jog up and down. I thought the filiment was jamming so I clicked on load filiment just to see what happens it extruded out just fine. Try the print and nothing comes out and the filiment jogs up and down again. I am at a total loss right now the print worked great last night and was almost done with out a single issue and now it won't print at all.

Note: abs 1.75 filiment, repetier host .92.10 hack de vinci 1.0, the idler wheel was a bit loose so I shimmed it and tighted it back up,I have removed the sticker off the front of the filament stepper motor and drew a black on the shaft. When I select the load filament option on the printer the black line spins and the filament extruds. when I select a print I tried one pre loaded on the printer and one downloaded from thingivers the black line no longer spins but rather jogs back and forth and the filament will not extrude. what could possiblely allow the stepper motor to function properly when selecting load filament and not to fuction when slected to print. I have adjusted the bed again and have lowered the bed just to see if it would spin. Its hard to think the motor is bad when it fuctions during the load filament. any thoughts.

I removed the extruder from the assembly and used the needle provided. I shine a light through the one end and could not see any signs of debris the light was not obstructed. The filament comes straight out no curling.

haven't had a issue in 40 prints and now this. I clean and level the bed every 4 prints. I feel that this graphix card driver glitch that happen is the root cause for this. everything whent to crap directly after that. It could be just perfict timing i guess.

This forum has been the best for help so far hope that there is a fix for this 

Thanks

luc-github, 2016-10-11:
If extruder works well on his own when you use load filament from printer can be :
1 - dirt in your nozzle preventing extrusion
2 - bed leveling issue, nozzle is too close to bed peventing extrusion to work properly
3 - z speed is too high when printing, so it miss several step and so bed is too close to nozzle
4 - issue with your slicer settings / temperature, so extrusion is not correct

PS: unlike your wrote this is not a forum, it is a bug tracker for firmware issue and it seems not the case based on your description

Chromewrecker, 2016-10-12:
Sorry I guess? But I do feel like this may be a software or bug issues. I have completely remove disassemble and throughly cleaned the nozzle leveled the bed multiple times slowed the z speed down checked all the settings and the temp. Like I said in the post the issue didn't come up until the graphics card error popped up and locked up the computer the 40 plus prints before this lock up had no issues now since the lock up it won't print. To me it seems like a software or firmware issue not a cleaning,speed or leveling issue. Everything is cleaner and more level then it was when it was delivered. Don't want to sound mad just frustrated at this hole thing what ever this video card error seems to have messed up the printer

luc-github, 2016-10-12:
From FW point of view it is not a problem for sure as it extrude normally when you use the printer menu so from printer side no issue. if you confirm that extrusion menu is OK, then you confirm printer HW/FW is fine.

Now from computer/software I have no idea of you configuration or what could be the problem

I suggest you to:
1 - heat/extrude from repetier, do some movement using manual control to check communication is correct
2-  review what GCODE is generated to check what temperature is used 
3 - try a simple cube to see if the issue does not come from the model itself

luc-github, 2016-10-19:
If no update I close issue

closed 2016-10-19 by luc-github