luc-github/Repetier-Firmware-4-Davinci #203: Doesn't build any arduino 1.5.8, 1.6.3, 1.6.8, 1.6.9

sasori7, 2016-10-13:
I've gone over and over with this and have followed the instructions, which seem to have maybe worked when the arduino version was 1.5.8.  However, all I have run into is verify errors at every turn.
Also, the current instructions 'out there' talk about selecting the arduino USB option, which is not present in any of the builds other than 1.5.8.
It seems to me that this (0.92) is being changed to support new arduino software, and even davinci versions - aio is now listed as option 4 in Configuration.h) but there are no version changes (like 0.93 or 0.9.2.1)

luc-github, 2016-10-13:
what version do you you use ? master or release?
Did you read carefully the FAQ ?

nicksears, 2016-10-13:
I've also been having issues and I've done this before (on 1.5.8 with an older firmware). I'll describe my issue in detail to see if I'm doing something wrong or if other have run into something similar.

I had originally installed everyhing on 1.5.8, but wanted to go back and unify all my configurations. (I added transistors to my 1.0s to control a chamber heater) so I started with the new master 0.92.10.

I first uninstall arduino and delete arduino15 from c:\users\(nick)\appdata\local (and roaming) to make sure I'm starting fresh. I then install arduino 1.5.9 as stated in the readme (not run from the extracted zip, does this matter?) install 1.5.8 sam boards from the board manager, and replace the variants and usbcore files (in c:\users\(nick)\appdata\local\arduino15...) when I open the firmware, select arduino due, and compile, with the default settings, everything looks great. A side note, I've heard of an error with recent arduino builds where you have to open arduino directly, then open files, not open an ino file directly or it will give errors when it compiles. However, when I change DAVINCI  to 2 or 3 I get

In file included from sketch\Configuration.h:185:0,

                 from sketch\Repetier.h:200,

                 from sketch\Commands.cpp:22:

sketch\pins.h:1425:140: warning: narrowing conversion of '128' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]

                         HEATER_0_PIN, HEATER_1_PIN, HEATER_2_PIN,FAN_PIN, E0_PINS E1_PINS E2_PINS TEMP_0_PIN, TEMP_1_PIN, TEMP_2_PIN, SDSS }

sketch\Commands.cpp:24:41: note: in expansion of macro 'SENSITIVE_PINS'

 const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS; // Sensitive pin list for M42

I'm able to fix this issue by changing line 24 in commands.cpp

const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS; // Sensitive pin list for M42

to int16_t instead (not sure if thats a thing) or just int. I know a little why this is the case, but couldn't explain it

I have another issue when I compile with REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1, but I'll post that here separately

luc-github, 2016-10-13:
the warning is not a big issue because it just to avoid to use in use pin with M42 command

luc-github, 2016-10-14:
one thing to help you to understand issue is not on FW side for compilation using default settings is: compilation is done automatically at each commit and result is display on readme:
[image]
if you click on it you will see compilation process log for every davinci model, with the status at the end
```
Sketch uses 204,220 bytes (38%) of program storage space. Maximum is 524,288 bytes.

travis_time:end:22236fc6:start=1475927195080407454,finish=1475927212755540349,duration=17675132895
The command "build_sketch ./src/ArduinoDUE/Repetier/Repetier.ino" exited with 0
Done. Your build exited with 0.
```

sasori7, 2016-10-15:
Thank you for your reply and effort.
I used master.
I finally gave up on the 'tutorials' that were out there and followed your instructions that are in the FAQ.
instruction #4 could be clearer; I looked for the variant.cpp file to replace and did some searching for the one to replace.  I didn't find it so I just put it in <PATH>\arduino\hardware\arduino.
So, it seems you can no longer just drop the master hardware folder into the arduino directory unless you're using 1.5.8; which is no longer compatible with your very helpful code.

luc-github, 2016-10-15:
I think you did not read the  FAQ
the answer is in it: https://github.com/luc-github/Repetier-Firmware-0.92/issues/166
https://github.com/luc-github/Repetier-Firmware-0.92/issues/114
and others

luc-github, 2016-10-19:
If no update I close issue

closed 2016-10-19 by luc-github

sasori7, 2016-10-20:
I would still polish step 4 so that it is more clear.  Also, increase
version numbers when a change is made.

luc-github, 2016-10-20:
feel free to submit your suggestions
about version, every big change is referenced by system version based on date, and number of build done during the day, it is in system version in settings menu.
0.92.10 is the original version from repetier, it just for reference