luc-github/Repetier-Firmware-4-Davinci #204: Z probing fails/errors

DaAwesomeP, 2016-10-16:
Whenever I try to auto level or try to use Z-min, the printer will move to the first spot and start to raise the platform to touch the probe. However, the moment it touches the probe, it errors. It's definitely detecting when it touches the bed/probe point, because it errors at that exact moment, This leads me to believe that the error is in software.

Da Vinci 1.0A

luc-github, 2016-10-16:
it works on my side ...
Remember z probing must be done after correct a manual bed leveling and a zmin at 0
if distance to probe is too big it generate an error by design.

You can use G31 command to check if have correct contact or not to get probe state

DaAwesomeP, 2016-10-18:
OK,
So I manually leveled again (this was the last time I touched the adjustment screws in this sequence). Then I tried Z-probe and auto level again. They still failed. I ran one of them again. It worked. I ran the auto level again. It failed. I run auto level again. It worked.

I really don't know what is going on...but it works, and it definitely helped, so I'm happy.

closed 2016-10-18 by DaAwesomeP

luc-github, 2016-10-18:
you need also to be sure plate and probe is really clean
I read also some people have to tight better the corner plates to not have probe failure as the plate/screw was little bit loose