luc-github/Repetier-Firmware-4-Davinci #205: Extruder & Extruder Stepper Motor Upgrade

SASpirit, 2016-10-19:
Hi Guys,

I have done the E3D V6 mod to my XYZ 1.0A and went want step further to replace the extruder with the Titan one which is supplied by E3D.  This has come out really good all compactly installed into the existing carriage.

**One question** - The new motor has the following notes:

http://wiki.e3d-online.com/wiki/Titan_Assembly

> The new motor has 400 steps/rev, giving higher resolution than the older one with 200 steps/rev. This must be taken into account when setting the firmware.
> The shaft on the newer model is slightly longer to allow for the use of thicker mounting brackets.
> In addition to the shaft being longer on the new model, the shaft flat does not extend as close to the motor face plate as the old model. This means that for thinner mounting brackets it may not be possible to tighten the gear on to the flat section, don't worry, as the grub screw will still cope with enough torque to run the the extruder when tightened on to the round portion of the shaft.
> 
> You can tell which motor you have by checking the model number or degrees per steps written on the sticker on the side of the motor:
> 
> New: 42BYGHM208P4.5-15-X2 & 0.9deg/step 
> Old: 42BYGHW609L20P1-X2 & 1.8deg/step 
 
Do I need modded firmware in order to cater for this change or is this something that can be driven via the UI setup?

Many Thanks
Sean

luc-github, 2016-10-19:
you just need to calibrate extruder steps I think, it is in EEPROM, about rotation you may change it in FW or change the wire to avoid to touch FW, but this depend of rotation

SASpirit, 2016-10-21:
Thanks for feedback.  I have found those setting and understand what needs to happen.  My only issue now is there seems to be an issue with the motor.  Either the wiring is off (which would be odd) or there is a power issue as the motor is simply making a noise and you see a vibration as it attempts to turn but it doesn't actually turn.  And this is with NO load on it at all.

luc-github, 2016-10-21:
this behaviour is typical of wrong wiring actually, try another combination

SASpirit, 2016-10-21:
Thank you, I will check that.

Last question:  How good is your math? :)

**Firmware Calibration**

Because Titan uses a known gearing ratio and a consistently machined drive gear, we can offer a fairly accurate steps / mm set-point.

> Steps per Unit (Extruder) = Motor Steps * Micro-stepping * Gear Ratio / (Hobb Diameter * Pi)

Standard motor steps / rev = 400 for the new style motor or 200 for the old style motor (Refer to assembly steps if unsure which motor you have).
*Standard micro-stepping = 16x
*Gear Ratio = 3
*Hobb Diameter (Effective) = 7.3

> 200 * 16 * 3 / (7.3 * 3.142) = 418.5
> 400 * 16 * 3 / (7.3 * 3.142) = 837

Will I be using **837** or is it a little different for the Davinci?

> To test your value, mark the filament 120mm from the extruder then extrude 100mm slowly. If you do not measure 20mm after extrusion has finished, you may need to tweak this value.

luc-github, 2016-10-21:
this value will be a theorical starting point but you must do the extruder calibration to tweak the step/mm.
and not I am not good with math, that is why I use computers to solve my issues

closed 2016-10-24 by SASpirit