luc-github/Repetier-Firmware-4-Davinci #206: Compile Error

MandrewM, 2016-10-30:
I just downloaded the latest version to re-flash my DaVinci 1.0. When Compiling the latest (16/08/29) version from the master branch there is an error in the HAL.h file. I have included a screen shot of the error I received. I will be re-flashing with an older version (15/06/04) that does not have this error.
[image]

luc-github, 2016-10-30:
Hi did you check the FAQ? looks like you use old version of IDE

MandrewM, 2016-11-01:
I will try updating my IDE to the new version.  I thought the problem was from the function SetInterruptPriority() not being defined elsewhere in Repetier. I guess this is built in to the updated IDE version? I am not very familiar with this code. Thanks!

luc-github, 2016-11-01:
it is a method from Serial which is arduino not repetier

Agurt, 2016-11-05:
Hello,

I seem to be having a similar problem except I'm using the correct version. Any help would be greatly appreciated. 
[image]

luc-github, 2016-11-06:
yes this different issue,  I guess happen after a module update, close ide and open again
if issue occur again remove your due module close ide open again and reinstall the  close ide then open again and  compi!e if ok then modify variant.cpp usbcore.cpp
if issue happen again uninstall everything and eventually delete due modules files manully
this is pure arduino issue not related to fw

Agurt, 2016-11-07:
Thanks for the help. A friend of mine came over and we ended up figuring out that IDE 1.6.9 isn't compatible with OSX Sierra. We switched to 1.6.12 and are able to upload fw to the board now.

I'm getting this error from repetier now though: 
`fatal:Heater/sensor failure Printer stopped and heaters disabled due to this error. Fix error and restart with M999.`

Could this have something to do with using IDE 1.6.12?

luc-github, 2016-11-07:
well if you use ide 1.6.12 and DUE 1.6.8 should not be the issue

looks like decoupled issue - but without any details it is hard to guess what is the problem

luc-github, 2016-11-16:
If no update I close issue

closed 2016-11-16 by luc-github