luc-github/Repetier-Firmware-4-Davinci #207: Add Semitec 104GT-2 Table for Davinci 1.0a models

zombiRon, 2016-11-24:
I've generated a thermistor table for the Semitec 104GT-2 as shipped with the E3D v6 hotend, this covers the original glass and the new cartridge type.

This table is for use with the 1.0a and any other Davinci models that have a single 1kohm resistor instead of the 4k7 and 10k as found on the 1.0 models. I know some users have removed the 10k and used the standard Repetier tables. You could swap out the 1kohm for a 4k7 and use the generic tables, but cut and paste is much easier.

Can this be added to the firmware as an option for users who have upgraded their hotends/thermistors please?

#define EXT0_TEMPSENSOR_TYPE 7
#define NUM_TEMPS_USERTHERMISTOR2 36
#define USER_THERMISTORTABLE2 {{284, 2400},{324, 2320},{371, 2240},{426, 2160},{491, 2080},{566, 2000},{655, 1920},{759, 1840},{880, 1760},{1020, 1680},{1181, 1600},{1363, 1520},{1566, 1440},{1789, 1360},{2028, 1280},{2278, 1200},{2532, 1120},{2781, 1040},{3018, 960},{3234, 880},{3424, 800},{3587, 720},{3719, 640},{3824, 560},{3904, 480},{3964, 400},{4007, 320},{4037, 240},{4057, 160},{4071, 80},{4080, 0},{4085, -80},{4088, -160},{4090, -240},{4091, -320},{4091, -400}}

luc-github, 2016-11-24:
Thank you , so this table is only valid for 1.0A/2.0A and AiO

Is it a measured table or calculated table ?

zombiRon, 2016-11-25:
it's a calculated table, I have attempted to verify with a temp probe on my multimeter but as you know multimeters aren't accurate over the range of temps we print at. However I have verified the resistance and voltage at each temp against the Semitec datasheet and these correspond to how the printer interprets the temp.

luc-github, 2016-11-25:
Ok will add this table in configuration.h with a define for easy selection

ZeekWorld, 2017-07-12:
Will the table above correct the temperature offset when installing a E3D V6 on a Da Vinci 1.0a?
I have done the upgrade and the temperatures showing on the screen are much higher than what the real temperature is. 
If so, where exactly do I need to paste the code?
Thanks!

zombiRon, 2017-07-12:
yes this is for use on a 1.0a, if you're using the supplied e3d thermistor. Open configuration.h add it into one of the user thermister tables I think 2 is empty. And define this as the thermistor earlier in the config. If you get stuck I'll post my config. Unless luc has added it as a pre defined (sorry luc I've not reviewed the code for a while)

ZeekWorld, 2017-07-12:
Thanks!, yes, I have been running the e3d thermistor and have not done any modification to the Da Vinci motherboard. I will definitely give it a try!

ZeekWorld, 2017-07-13:
I am definitely stuck!
zombiRon, if you want to post your config I would appreciate it so I can review it and have an idea of where exactly to paste the code.

ZeekWorld, 2017-07-13:
I think I figured it out.

ZeekWorld, 2017-07-13:
I figured it out how to paste the code and unfortunately the table doesn't work right. With the type 7 and userthermistor2 temperature on the extruder shows 0. Tried to load filament, temperature stays at 0 when "heating". I'm going back to Table 5 for now until you guys can test and confirm this is a working code.

I've attached some screenshots.
https://www.dropbox.com/s/vvz6ls8r1y22sfz/Image%20323.jpg?dl=0
https://www.dropbox.com/s/x6wrk3guodhbibl/Image%20324.jpg?dl=0

https://www.dropbox.com/s/7mvczffuzi7lo0a/20170712_190721_resized.jpg?dl=0
https://www.dropbox.com/s/cuwlmt8e8duj5bw/20170712_190736_resized.jpg?dl=0

luc-github, 2017-07-13:
@zombiRon no it is not yet integrated still added the #Waiting to be processed label - I have been quite busy lately sorry

luc-github, 2017-07-13:
@ZeekWorld did you managed to use the table ?

zombiRon, 2017-07-13:
If you look at the code, there is a if MODEL==0 clause before where you have pasted the table

Look a little further down and paste it into the following else section. You have a model 1 device as far as repetier is concerned, so its tying to use an empty table. 

Hope this helps. 

Regards,
Aaron

ZeekWorld, 2017-07-13:
@luc-github I am going to give it another shot tonight.
@zombiRon Thank you for the explanation. I believe after trying last night I realized per your message I was missing the #if model==1 you described.

I will flash the Da Vinci tonight and let you both know how it goes.

**Note:** Something I found out after hours of reading and looking for more information is that the original e3d V6 has the Semitec 104GT2 thermistor which @zombiRon created the table for, but I may have purchased a clone of the e3d V6 which most of the threads online refer to a NTC 3950 thermistor with very similar characteristics. So I am not sure if having a different thermistor will make any difference even both are 100k.

zombiRon, 2017-07-13:
This table will not work with the NTC thermistor, the tables are very specific to each thermistor. Table 5 is the closest you'll get to the generic NTC thermistors that come with the clone hot ends. 

Regards,
Aaron

ZeekWorld, 2017-07-13:
Thanks Aaron.

reklawdiov, 2017-10-06:
I just installed my new style e3d v6 hotend with the cartridge style thermistor, I got it from an authorized dealer so it's not a clone. I used @zombiRon 's custom table and when I set the extruder temp in the LCD menu to 265c it never gets there, it stops at around 257c. it's definitely hot enough to feed and melt ABS all day long but I'm not convinced it's reporting the correct temp and I can't verify with my temp gun because it wigs out with reflections. Am I missing a setting? PID tuning? 

where are these resistors that I need to check for 1k or 4k7 or 10k? I'm proficient with soldering so If I can bump off or replace them to use the standard table that's cool. Also, this is a 1.0A printer.

zombiRon, 2017-10-06:
Hi reklawdiov,
The 1k internal resistance on the 1.0A is not as easily accessible as it was on the 1.0 version. I attempted to track it down, after hours of following traces I couldn't find a simple way of physically changing the internal resistance. However, while building the custom table I realised that the 1K makes more sense from a calculation point of view. I think your issue is down to software configuration. Can you post your config.h please? But just to be sure, disconnect the thermistor and measure the resistance on the printers thermistor pins.

reklawdiov, 2017-10-06:
@zombiRon  I'll check with the multimeter when I get home after work but here is my config.h file, thanks for checking!
[Configuration.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/1363153/Configuration.zip)

zombiRon, 2017-10-06:
Your config is good, only other thing I can think of is your EEPROM settings, here are the relevant settings from my EEPROM...

Temp Ext PLA: 180.000 
Temp Ext ABS: 220.000 
Temp Bed PLA: 60.000 
Temp Bed ABS: 90.000
Extr.1 heat manager [0-3]: 1
Extr.1 PID drive max: 255
Extr.1 PID drive min: 40
Extr.1 PID P-gain/dead-time: 7.0300 
Extr.1 PID I-gain: 0.7000 
Extr.1 PID D-gain: 17.6900 
Extr.1 PID max value [0-255]: 255
Extr.1 temp. stabilize time [s]: 0
Extr.1 temp. for retraction when heating [C]: 200
Extr.1 distance to retract when heating [mm]: 0
Extr.1 extruder cooler speed [0-255]: 255
Extr.1 advance L [0=off]: 0.000

reklawdiov, 2017-10-13:
@zombiRon that seems to have done it! I finally got a chance to test and tweak today, it was using "slowbang" heat manager 3 and the PIDs were way off. I've updated and am able to hit 270C without a problem (so far) other than the menu maxes out at 270 instead of 280 that's set as max in the printer settings. probably for the best since it says not to go above 285 I think for the thermistor.

jnowak1054, 2019-10-30:
Please excuse my ignorance as I'm probably not thinking clearly and missing something simple. Currently running Repetier 92.10 on a davinci 1.0a. I've been attempting to build a proper table for my unknown thermister in my current hotend as it reads way low (reads 280C @ 240C as measured by a thermocouple.) 
When I change line 365 in Configuration.h 
#define EXT0_TEMPSENSOR_TYPE 5 
to #define EXT0_TEMPSENSOR_TYPE 7
the printer reports back 0C
I tried pasting a new calculated table into #define USER_THERMISTORTABLE0 on line 723, #define EXT0_TEMPSENSOR_TYPE 5, with no change in reported temperature from the printer in original config.
Compared the above config.h to my own and see line 721 also has #if MODEL==0, and tried changing that anyway as well. Tried along with different combinations of EEPROM Extr.0-3 heat manager [0-3]:
I'm obviously missing something. Any help or a point in the right direction to look or read would be greatly appreciated.

luc-github, 2019-10-30:
@jnowak1054 you have a 1.0a so your model must be MODEL 1
keep #define EXT0_TEMPSENSOR_TYPE 5

you just need to modify the lines : 
https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L737-L738

jnowak1054, 2019-10-31:
Thanks! Now that I see it, I also see my question was already answered above. Thanks again!

zombiRon, 2019-10-31:
This reminds me, I've not been active for a while; luc, your corexy code works too. I converted my davinci to a custom corexy. Happy to share the drawings if anyone is interested.

sisensee, 2019-10-31:
Always interested!

zombiRon, 2020-06-22:
I'll be putting it on thingiverse but here's a sneak preview to my DaVinci CoreXY conversion. https://a360.co/3dlKGY9

closed 2020-12-13 by luc-github