luc-github/Repetier-Firmware-4-Davinci #208: Home X

wise990, 2016-12-08:
I find that when I home the X axis either on its own or a by a home all that I can no longer manually move any axis in any direction.  I did find somewhere else and maybe a completely different issue that by issuing M100 that manual movement is restored.  I think this home issue may also be affecting my end gcode because the x and y home but the bed never actually moves down after the print.  My end gcode is below

M117 Done
G92 E0 ; reset extrusion distance
M104 S0 ; turn off extruder
M140 S0 ; turn off Bed
G90 ; Absolute
G28 X0 Y0 ; home
M117 Down Bed
G1 Z200 ; Lower bed
M117 Cleaning
M100 ;Cleaning
M84 ; Disable motors
M117 Cooling
M190 S55 ; Bed (cooling)
M116 ; Wait for all temperatures
M140 S0 ; turn off Bed
M117 Pickup piece
M300 s2500 p400 ; Beep
M300 s2000 p400
M300 s1500 p400
M300 s1000 p400
M84 ; Disable motors
M117 Printer Ready

It seems like an error of X3 is reported but I cant find any information as to what that might mean anywhere.  Any help would be great.

luc-github, 2016-12-08:
can you clarify :  
> when I home the X axis either on its own or a by a home all that I can no longer manually move any axis in any direction
I cannot reproduce using printer menu, do you have detailed steps to reproduce it ?

also what is the : 
>error of X3 is reported
do you have the error message or log ?

wise990, 2016-12-08:
So I first noticed this after running a print job.  The first thing i noticed was that the bed never traveled down as instructed in my end gcode.  Then when I went into the manual tab in RH I was unable to click to move in any direction.  I can home and of the axis but I cannot move them by clicking the arrow.  If i turn of the machine and back on then reconnect to RH I am able to move again until I home X (homing Y and Z do not cause the issue)  When I am locked up the only thing that comes through in the log is X3.  When I home X I see an error message of selectExtruder:0  Maybe i am looking at the wrong place for the log.  I click the little bubble in the log window that says errors.  Any instruction to move just shows X3 and nothing moves.  If I send M100 (clean) then all work again until I home all or just home X.  It is like the manual control gets locked out, i do see that is trying to send commands though nothing moves.  I have not had repetier on this machine very long and I dont think this has always happend but  I am not sure what I would have changed to make this now occur.

luc-github, 2016-12-08:
what version of FW do you use ?
what version of repetier do you use ?
what OS do you use ? 
what baud rate do you use ?

wise990, 2016-12-08:
FW .92.10
Repetier Version 1.6.2
Windows 10
Baud 230400

wise990, 2016-12-08:
Here are the commands, error and ACK of a manual click move where the printer does not move. 

14:39:49.128 : N174 G91*19
14:39:49.128 : N175 G1 Y50 F4800*61
14:39:49.128 : N176 G90*16
14:39:49.128 : ok 174
14:39:49.129 : ok 175
14:39:49.129 : x3
14:39:49.136 : ok 176

luc-github, 2016-12-08:
G9119 ?
G1 Y50 F48861 ?

these values are crazy !!! You may have USB communication problem

wise990, 2016-12-08:
I think something didnt copy right when i pasted the log.  It looks like it removed * so it is really G91*19 and F4800*61

wise990, 2016-12-08:
It seems to always remove the asterisk from the gcode i paste in the forum

luc-github, 2016-12-08:
ok so to reproduce issue it need : 
1 - print use your end gcode
2 - go to repetier manual tab then home X
3 - use X direction arrows in manual to move X axis

correct ?

wise990, 2016-12-08:
I dont even need to do the end g code.  I fire up the printer and go the manual tab and then home X now using the directional arrow for any axis will no longer move the printer, I can still home any axis, just cant use the arrows.  (M100 will get the arrows working again)

luc-github, 2016-12-08:
I will install repetier1.62 tomorrow and test

 last question what is your printer model ?

wise990, 2016-12-08:
davinci 1.0

I am not sure it is a repetier problem.  I dont think i can move the axis with the buttons on the machine when this lock up occurs, I would have to confirm that again tonight, but i am pretty sure nothing I did would move the printer.

TGMods, 2016-12-09:
I have a Davinci 2.0A and I use the following End G-Code:

G1 X0 Y0 Z200 F1000 ; home X and Y axes and drop bed
M103 ; turn off extruder and retraction
M104 S0 T0 ; turn off left extruder
M104 S0 T1 ; turn off right extruder
M140 S0 ; turn off bed
M107 ; turn off fan
M127 ; turn off valve
M84 ; disable motors

I found that I had to run the G1 line before anything else or it wouldn't work. 

I think once the motors are disabled the firmware forgets the HOME position, at leased that's what happens on my Core XY printer which uses Marlin.

luc-github, 2016-12-09:
Yes if M84 or automatic disable motor will reset the home flags by safety, that is why I have added a message on FW if home flag is not set - it will display on screen if you use printer menu a message " the position is unknown - do you want to home first ?"

wise990, 2016-12-09:
I am able to get the end GCode to do what I want but I still have the issue where a home (g28) will cause the motors to no longer respond.  I can cause the lock up buy only using the manual tab and the home buttons on there.  Once in a while i see a message on the printer that "Steppers Are Disabled" but I have no clue what causes that or how to reenable with gcode

wise990, 2016-12-09:
I am able to get the end GCode to do what I want but I still have the issue where a home (g28) will cause the motors to no longer respond.  I can cause the lock up by only using the manual tab and the home buttons on there.  Once in a while i see a message on the printer that "Steppers Are Disabled" but I have no clue what causes that or how to reenable with gcode

wise990, 2016-12-09:
Is there any way to tell what is blocking a command?  I assume something about homing causes more manual commands to be blocked for some reason.  Here is a video I made showing the steps to recreate [https://youtu.be/8LImPtoWEJE](url)

luc-github, 2016-12-09:
When steppers are disabled  doing a home or movement command reneable them for the axis called

I have tried several time and cannot reproduce issue
I did same sequence as you in video using the arrows of manual control :
Home Z
Y +50
Home Y
Y +50
Home X
Y+50
And all is ok still can command axis

Does the printer menu allow you to move axis when you are stuck ?

Also do you use ping pong ? in repetier host settings ?

wise990, 2016-12-09:
I am unable to move with the printer menu when i am stuck.

I am not using ping pong.  Could the cache size be an issue?  I have changed that to 202.

How would i know that i have a command that is blocked?

Thanks for your help on this issue.

luc-github, 2016-12-09:
please use ping pong it is recommended to not overload buffer and set buffer to 63 
when command is blocked repetier said waiting for Ok but only if you use ping pong

wise990, 2016-12-09:
I have changed to ping pong and set buffer to 63 but still have the same issue.

Here is the log for a failed move

17:19:17.075 : N53 G91*39
17:19:17.075 : N54 G1 Y50 F4800*15
17:19:17.075 : Echo:N53 G91
17:19:17.075 : N55 G90*32
17:19:17.075 : x3
17:19:17.075 : Echo:N54 G1  Y50.00 F4800.00
17:19:17.075 : Echo:N55 G90

x3 seems to be the error line

wise990, 2016-12-09:
home all when issued right from the printer also locks it up.  I dont even need to use RH to recreate the issue.

luc-github, 2016-12-10:
you mean printer freeze ?

I cannot reproduce - and this bug is pretty straight - I would saw it as I only use printer menu

I think somehow the FW has issue - I would suggest to flash it again

Did you had issue to flash it before ?

wise990, 2016-12-10:
I havent had the printer very long and I flashed to repetier firmware very soon.  I will try to reflash firmware.

luc-github, 2016-12-29:
any update on this ?

wise990, 2017-01-03:
honestly I just stopped moving the printer around in a fashion that replicates this issue.  I didnt have time  to reflash the firmware.

luc-github, 2017-01-03:
No problem - close issue - just open again if need

closed 2017-01-03 by luc-github