luc-github/Repetier-Firmware-4-Davinci #209: Bed temperature wrong

Hannes54, 2016-12-10:
Hello
Since I use on my DaVinci 2.0A FW0.92.10 I have a temperature difference of about 25 ° C at my print-table. The display will show too much on the display and also in the Repetier-Host 25 ° C.
I have checked with an infrared thermometer and with several temperature sensors.
How can I correct this?
Everything else works very well! Thank you!

Greeting

Hannes
P.S. I use Google translator because my english is bad.

luc-github, 2016-12-10:
the temperature table used for the 2.0a is the one measured by 2.0a owners, you need to do your own table if it does not work

the github has the excel file with the measurement done, you can check
you mean you have a diference  of 25 degres at each temperature ? in more or in less?

Hannes54, 2016-12-10:
Thank you for the quick answer.
at Roomtemperature it is ok but at 100°C in Display are only 75°C on table. At 120 so are 95 on table. 
Because i am printing ABS i have problems. 
How can i make my own table?
Sorry i speak german.
Greetings and thanks for your great job! It is like i have a new printer after flashing.
Hannes

luc-github, 2016-12-10:
you need to build a matching table of ADC value and 8 X temperature in celcius 
ADC value get with M105 X0, and your measured temperature , then replace the one here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L738-L739

Hannes54, 2016-12-10:
thank you very much!
i wll try it but it is hard for me.
what means ADC value?
where i can set "M105 x0" ?
if i make changes there i have to flash again?

luc-github, 2016-12-10:
ADC is analog raw value measured by sensor
you need to connect repetier host and use manual tab or use arduino serial monitor to send command
yes you need to reflash as you modify FW

Hannes54, 2016-12-10:
once more, thank you

Hannes54, 2016-12-10:
If i heat the table (with Rep Host) to 100° and use M105 x0 i receive: 106 / 28
I do not understand this values.
Sorry it is not easy for me to understand this.

luc-github, 2016-12-10:
Long time I do not use repetier host - I forget it filter temperatures output sorry, using arduino serial monitor the output is like this you see the temperature and the RAW value 
`T:25.41 / 0 B:29.91 / 0 RAW2:3614 B@:0 @:0 T0:25.41 / 0 @0:0 RAW0:2690 T1:26.22 / 0 @1:0 RAW1:2687`

Hannes54, 2016-12-10:
sorry, if i use serial monitor in IDE with M105 x0 (at 100°C) then i become : resend 627
If i change Temperatur it is the same. endless "resend 627"

Hannes54, 2016-12-10:
Hello,
now i have found that in S3D it works with M105 x0. i can see the values. why it does not work with serial-monitor in Arduino-IDE? 
tomorrow i will try to change the user-table.
what are the values for the print-table? RAW 0 , 1 or 2 ?

thank you for gret support!

greetings
Hannes

Hannes54, 2016-12-11:
Hello,
now it works to with serial monitor. 
i found in config.h 3 usertables 0,1 and 2. wich one is for bed?
how i must input new values? is anywhere a step by step instruction?

Sorry but i have no idea how to correct my temperatures.
Hannes

luc-github, 2016-12-11:
please read again : https://github.com/luc-github/Repetier-Firmware-0.92/issues/209#issuecomment-266233134

Everything is there

Hannes54, 2016-12-12:
Hello Luc,
now i made my own usertable. it works fine!
i dont know why but the values are much different from original.

thank you very much for your help and sorry for my stupid questions.
do you need my results?

with kind regards
Hannes

luc-github, 2016-12-12:
Good - measurement results are affected by several factor - weather / humidity / sensor position temperature stability so I am not surprised you may got different results - but the key point is : it meet your needs, so all good

Yes please share : as original table come from 1.0a owner I cannot say it is bad as it is used for one year already - but the need could come from some hardware changes so, it could affect others people

Just copy the table here - I will have a look and can be helpful for someone elese in same situation as you

Hannes54, 2016-12-12:
Here is the exel-file. 
[Temperaturen.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/646303/Temperaturen.xlsx)
it is the file what i found and i added my own. (Hannes)
may be  there are better ones but for me it is ok.
the temperatures where measured on surface of print-bed with pt100 sensor.
values under 18°C and about 120°C are fictive!

luc-github, 2016-12-29:
Thanks for sharing I will add as option in tables for 1.0A/2.0A/AiO

HorstBaerbel, 2020-01-14:
Is there a way to calculate the ADC values from the resistor values. I have removed my (default DaVinci 1.0A) bed temperature sensor to measure and compare it to a DS18B20. I have the resistor values for a couple of temperatures and want to convert these to a custom table. Any hints?
[image]
Or should I use USE_GENERIC_THERMISTORTABLE_1, calculate a beta and use that? I guess I'd need to unsolder some resistors then?

luc-github, 2020-01-15:
R29 is for Extruder not Heatbed

I suggest you to do your own table by doing measurement - as you can see in https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/bed-extruder-tables.xlsx
depending of people for same heat bed values may vary due to environement and cheap quality of Davinci components. 
several people submited table based on calculation for extruders and none of them were working for 100% people - so better to do your own IMHO

HorstBaerbel, 2020-01-15:
Thanks for the answer. Yes, I meant R29 for the hotend, R28 for the bed, sorry for the missleading wording. I've already found the xlsx file, but I measured the thermistor by removing it from the printer to compare it to a DS18B20.
As there seems to be no way to calculate the ADC values, I guess then I need to re-measure with the thermistor inside the printer and polling the ADC values using "M105 x0" :/

HorstBaerbel, 2020-01-15:
Worked like a charm. Now my bed temperature is max ~1°C off, according to a DS18B20 clamped to the glass bed with some thermal paste in between. This should definitely have a hint on the start page! My bed temperature was of by more then 10°C. This causes lots of problems...

> you need to build a matching table of ADC value and 8 X temperature in celcius
> ADC value get with M105 X0, and your measured temperature , then replace the one here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L738-L739
> 
> and here the references currently used: https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/bed-extruder-tables.xlsx

Note that per default USER_THERMISTORTABLE1 is used for the bed (because of [#define HEATED_BED_SENSOR_TYPE 6](https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L840)), which is defined in [Line 741-742](https://github.com/luc-github/Repetier-Firmware-4-Davinci/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L741-L742), not 738-739.

HorstBaerbel, 2020-01-15:
Here's the new table (including Hannes' values): [bed-extruder-tables.xlsx](https://github.com/luc-github/Repetier-Firmware-4-Davinci/files/4065836/bed-extruder-tables.xlsx)

HorstBaerbel, 2020-01-16:
I've now also measured my extruder with a Type-K thermocouple attached in the hole where the extruder sensor is. Temperatures were off, but not by much (more or less ~5°C). Here's the updated table (including Hannes values), including my extruder values: [bed-extruder-tables.xlsx](https://github.com/luc-github/Repetier-Firmware-4-Davinci/files/4070540/bed-extruder-tables.xlsx)

MasterCATZ, 2020-08-01:
tested my da vinci 2.0 dual fan  today and temp is ~5+- with lazer pointed at it 
a lot better than last time a few year back when it was only doing 60deg with 100deg set

closed 2020-12-13 by luc-github