luc-github/Repetier-Firmware-4-Davinci #210: Two black bars and computer will no longer recognize printer

Chromewrecker, 2016-12-11:
So I have had a sires or issues with both my printer and computer just got my computer back up and running and now I'm trying to get the printer back up but seem to have a issue. After arduino uploads the repetier host the printer does not do anything the black bars remain and the computer will get an error stating that the what is plugged in the USB port is not recognized. I have tried a couple of times same thing keeps happening. I can reset the printer and the message will go away until I try to upload the program again any thoughts out there

Chromewrecker, 2016-12-11:
And no I can seem to get the printer to wipe it keeps saying USB device not recognized

Chromewrecker, 2016-12-11:
Jumping j37 no longer seems to make a difference

luc-github, 2016-12-11:
for the black bars please check FAQ
for the computer detection 2 possibilities:
1 - issue with computer port,  restart computer, try several ports
2 - the printer has hardware issue

RootWizard, 2016-12-28:
@Chromewrecker Please see my comment in https://github.com/luc-github/Repetier-Firmware-0.92/issues/213  I had a similar issue and fixed it.

luc-github, 2016-12-29:
if no update I close issue

closed 2016-12-29 by luc-github