luc-github/Repetier-Firmware-4-Davinci #211: Davinci 1.0a hangs at 79% not matter what I do.

TK1362, 2016-12-11:
I flashed my davinci 1.0a via the J37 two black bars and followed every tutorial in existence. Arduino 1.6.9 Due 1.6.8 copied the hardware folder ........gets to 79% and just stays there forever?

luc-github, 2016-12-11:
did you checked the FAQ ?  I guess you are at 79% of verification and not upload,  https://github.com/luc-github/Repetier-Firmware-0.92/issues/185

TK1362, 2016-12-11:
I did check the FAQ, I've tried different cables and ports but still same effect and the 79% is during the verify process

luc-github, 2016-12-11:
so as mentioned in FAQ, disable verify process

TK1362, 2016-12-11:
Thanks, I'll try it !!!

TK1362, 2016-12-11:
I disabled the verify process in arduino and it completed but now the printer is not recognized?

TK1362, 2016-12-12:
I tried it again and it worked! Thanks for the help luc..........you the man!

luc-github, 2016-12-12:
good, happy printing 😄

closed 2016-12-12 by luc-github