luc-github/Repetier-Firmware-4-Davinci #213: Endless Boot loop

zenvent, 2016-12-25:
Hello, 
I successfully flashed the Davinci 1.0 printer I have a long time back. It was having some issues so I decided to reflash today.

I am able to connect and flash the printer, It uploads and claims verify is successful. The printer does not do anything afterwards though, it remains dark with two black bars. I can hear a little tick ever few seconds, which usually indicates the arduino is power cycling.

I've tried flashing a couple times, and tried cold booting the printer as well no luck. If I short JP1 again the printer stops making the 'booting click' noise, so it seems I can still clear the eprom.

Davinci 1.0 (JP1)
Arduino 1.6.9
Due Board 1.6.8

Any ideas?

luc-github, 2016-12-25:
Did you check the FAQ ? - there is one about black bars

zenvent, 2016-12-26:
I had checked the FAQ, variants.cpp was already taken care of. I ended up reverting to 1.5.8 arduino, using an older revision here, and power cycling the printer after firmware upload. That seems to have done it.

zenvent, 2016-12-26:
Though I have successfully reflashed the printer, I still have the same issue. It will not talk to my octopi server, I've tried two different PIs now, older and new versions of octoprint and different baudrates (settings changed on printer)... no luck fixing that yet.

luc-github, 2016-12-26:
variants.cpp is in different location depending of which version of ide you are using.
For octopi I cannot help as I do not use and do not know
did you tried when you connect to arduino terminal you see the printer sending message or use repetier host ?

RootWizard, 2016-12-28:
@zenvent Are you 100% sure you updated the variant.cpp in C:\Users\<name>\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x
with the one from the repo in Dir-to\Repetier-Firmware-0.92-master\src\ArduinoDUE\AdditionalArduinoFiles\Arduino - 1.6.9 -Due 1.6.8\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x
and the USBcore.cpp from the same dir in .../cores/arduino/USB/USBcore.cpp?

All you need to do is copy the Arduino15 folder from the repo into AppData dir and say yes to overwrite the USBcore and variant.cpp. Then resume flashing as before. I say this because it is slightly different from the 1.5.8 method, especially if you use the portable arduino package like me.

[image]

luc-github, 2017-01-06:
Ok if no update I close issue

closed 2017-01-06 by luc-github