luc-github/Repetier-Firmware-4-Davinci #214: I cant flash my new davinci 2.0

bbozzi12, 2017-01-05:
so, last year i bought a 1.0a and i flashed it with repetier, no problem. I got my new 2.0 in the mail today and i have tried just about everything possible to flash it with repetier. I have tried almost every version of arduino (1.5.8 , 1.6.8 , 1.8.0), and i have tried replacing the variant.cpp and hardware files, but still error messages when compiling or uploading with black bars. I would love to get printing soon soon but, this is really giving me a hard time.

luc-github, 2017-01-05:
without any description of what you do and  what issue you meet it is hard to say what you do wrong 
I suggest you to remove everything related to arduino from your PC and re do process from scratch, check each step using FAQ before going to next one, stick to IDE 1.8.0 with Due 1.6.8 to avoid compilation issue and focus on your flashing problem related to variant.cpp I guess

bbozzi12, 2017-01-05:
alright i will give it a shot

bbozzi12, 2017-01-05:
ok so I have used 1.8.0 and I replaced the hardware file, I compiled and it worked just fine. But when i upload i get the error: processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:211)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)
Caused by: processing.app.helpers.PreferencesMapException: upload.pattern
	at processing.app.helpers.PreferencesMap.getOrExcept(PreferencesMap.java:267)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:205)
	... 5 more

luc-github, 2017-01-05:
well this is arduino issue not FW what is your OS ?
did you try to unplug usb and replug and retry ?

bbozzi12, 2017-01-05:
I use windows 10, and yes I have unplugged my usb and tried again.

luc-github, 2017-01-05:
well looks issue with preference file so delete it when arduino is closed
it is located at: `C:\Users\<your loggin name>\AppData\Local\Arduino15` 
then open arduino and retry
try also to select right port

bbozzi12, 2017-01-05:
i deleted the file and selected the right port and board. I compiled and i got the sensitive pins error, then i got the same previous error for uploading the FW.

luc-github, 2017-01-05:
sensitive pins message is not an error it is a warning 

did you power down the printer and the PC and retry connecting to another usb port ?

bbozzi12, 2017-01-05:
I just tried plugging the printer into the back of my computer on a different port and  i got the same error message.

luc-github, 2017-01-06:
Ok,
I cannot reproduce your issue (java error message) with my laptop windows 10 ide 1.8.0/ due 1.6.8 
but flash was very slow and result was black bars and needed to use jumper again to get printer detected
but flasing with my desktop windows 7  ide 1.8.0/ due 1.6.8 flash was fast and no issue, this is the one I always use

So it can be :
1 - window10 /arduino 1.8.0 ide incompatibility - I will chek tomorrow by reverting to 1.6.9 ide on the windows 10 laptop as it is 1am here so not ready to do it now
2 - an hardware issue if I got same result with ide 1.6.9 on windows 10 as I never flashed with this laptop before

will feedback tomorrow

bbozzi12, 2017-01-06:
well i have an update. so i updated my board to the 1.6.9 version and i got the project to upload but it still shows the black bars on my printer, any suggestions?

bbozzi12, 2017-01-06:
im going to attempt to flash with another pc, i will get back to you

luc-github, 2017-01-06:
after updating to 1.6.9 did you update variant.cpp ?
what is the path of the variant.cpp you update ?

bbozzi12, 2017-01-06:
i did not update the variant.cpp

luc-github, 2017-01-06:
well that explain the black bars then

bbozzi12, 2017-01-06:
I got it, thanks for the help man. You should probably get off to bed now.

bbozzi12, 2017-01-06:
i got my printer flashed but i have one last issue, i will open a new thread.

luc-github, 2017-01-06:
On my side I realized I forget to update the variant.cpp as the previous installation was updated, so actually
my laptop windows 10 ide 1.8.0/ due 1.6.8 can flash without problem
Note to myself : do not try anything after midnight , Murphy law will be active at 200%

@bbozzi12 how did you solve your issue ?

bbozzi12, 2017-01-06:
i finally got it flashed by reinstalling the FW and Arduino 1.8.0, then i only replaced the variant.cpp file in the hidden directory and it worked.

luc-github, 2017-01-06:
Ok so you followed the readme then :smile: 
Be careful because you did not update usbcore.cpp, you must not unplug your printer when connected to repetier host, it will freeze printer, if watchdog is working properly it will restart printer, but if not, if printer is heating it will be heating out of control - so watch out

closed 2017-01-06 by luc-github