luc-github/Repetier-Firmware-4-Davinci #215: Error: isPositionAllowed, Need Help.

bbozzi12, 2017-01-06:
i got my printer DaVinci 2.0 DUO flashed with repetier after alot of trouble, and when I connect to repetier host i can move the extruders around, in the bottom log when i try to manually move the print head it says: Error: isPositionAllowed
X: NAN Y: NAN Z: NAN E: NAN
any suggestions?

luc-github, 2017-01-06:
I do not reproduce it :
[image]

Did you do the M502/M500 command to properly setup virtual EEPROM ?

bbozzi12, 2017-01-06:
no, how do i do that?

bbozzi12, 2017-01-06:
i managed to fix it by loading my failsafe, i forgot to do it, and you were right. It is 4am here and I am not able to do anything right. Thanks for all the help man.

luc-github, 2017-01-06:
yes loading failsafe and save them do the same 
Happy printing

closed 2017-01-06 by luc-github