luc-github/Repetier-Firmware-4-Davinci #216: Davinci 2.0A Help

T-RexCAD, 2017-01-06:
Flashes fine, but nothing works
DaVinci 2.0A  - Jumper J37, #define DAVINCI 3 and #define MODEL 1
It seemed to flash perfectly, all commands on printer screen seem fine and saved.
Now the commands on the printer screen will not do anything.
e.g. unable to move x,y or Z, home doesn't work, wont heat extruder, loads sample print then stops. 
I re read the instructions and re flashed, but still no change.
The printer worked as well as can be expected on stock firmware before the flash.
I am stumped whats wrong

luc-github, 2017-01-06:
Did you do the load fail safe or use the command M502/M500 as mentioned in readme ?

please do not use print sample from SD card, they are for XYZ software, they won't work with repetier as no heat command on sample files

T-RexCAD, 2017-01-06:
Thanks, I didnt do that.
I have now installed Repetier Host, but cannot find the M502/M500 files or how to load them

luc-github, 2017-01-06:
line 10 in readme:

**After printer restarted do not forget to send G-Code M502 then M500 from repetier's Print Panel tab or from the printer menu "Settings/Load Fail-Safe" and accept to save the new eeprom settings.**

T-RexCAD, 2017-01-07:
Thanks for the help
Yep the problem was  failing to send M502 then M500 into printer.
After sending and setting the height, the printer worked perfectly
Printing now.

closed 2017-01-07 by T-RexCAD

luc-github, 2017-01-07:
Good - happy printing 😄