luc-github/Repetier-Firmware-4-Davinci #218: How to use Extruder #2 ONLY in repetier?

bbozzi12, 2017-01-08:
i am using different filament colors, and i cant seem to find a way to use only the 2nd extruder.

luc-github, 2017-01-08:
like any other printer: T0 is extruder 1 and T1 is extruder 2
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L63-L64

So you need to configure your slicer to use T1, this is not specific to Davinci

bbozzi12, 2017-01-08:
do i need to manually set that in the g-code?

luc-github, 2017-01-08:
it depend of your slicer capabilities - some have macro some don't, some do not support more than one extruder, some automate the process

bbozzi12, 2017-01-08:
how would i do it in cura?

luc-github, 2017-01-08:
I do not use cura since a long time but you can check the settings I shared https://forum.voltivo.com/showthread.php?tid=8253&pid=2459#pid2459

luc-github, 2017-01-14:
if no update I close topic

closed 2017-01-14 by luc-github

Shawnh77, 2019-03-12:
Any chance you can share your settings again.  it appears that the forums have been shutdown due to spam.

luc-github, 2019-03-12:
I do not have them anymore sorry - but I think cura was updated several times so I am not sure they would be still usable