luc-github/Repetier-Firmware-4-Davinci #219: No Com after flash

cozmicray, 2017-01-09:
Flash with IDE 1.8.1 latest RF 0.92 
 to DaVinci 1.0 (with Vstruder extruder / Delta printr mini hotend)
COM7 shows post jumper connect
Post compile / upload
No COM   USB device problem

What am I doing wrong?

luc-github, 2017-01-10:
I never tried version 1.8.1 so I cannot say - is printer working well after flashing ?

cozmicray, 2017-01-11:
Since it does not connect to a COM  --- the printer does nothing
LCD screen blank  two black lines  before and after flash
I think LCD has bad ribbon cable?
Should I try IDE 1.5.8?

luc-github, 2017-01-11:
this is typical issue with variant.cpp - please check FAQ https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+
and please stick to readme version: use ide 1.8.0 with Due 1.6.8

cozmicray, 2017-01-12:
I have Arduino ide 1.8.1 and Due 1.6.11
are you saying I have to use 1.8.0 and Due 1.6.8
Which FAQ should I explore?

Post flash  shows up as ----  Bossa Program Post(COMX)
Is there any firmware to test my controller board communications chip

luc-github, 2017-01-13:
read all readme carefuly
read all FAQ I gave in previous post, they are the same mentioned and linked in readme if you read readme

your issue is you use wrong variant.cpp so it means: you do not read the readme and do not follow the instructions (seems the case as you use ide 1.8.1 and Due 1.6.11) and/or you do not update properly the variant.cpp and it is explained in FAQ

cozmicray, 2017-01-13:
Would be nice if you stated what variant.cpp is required
where I can get it, where it goes in my arduino files

All my other arduino stuff seems to like 1.8.1
and finding your exact load of IDE and libraries impossible

Thanks

luc-github, 2017-01-13:
I did in readme

cozmicray, 2017-01-13:
Thank you  -- Finally got it to work
you can only have one variant.cpp  --- renaming old one doesn't work
variant.cpp must go in special place for arduino install (seem to work for me)
C:\Users\xxxxxx\AppData\Local\Arduinoxx\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x
variant.cpp has date of 12/31/2016

Thank you

luc-github, 2017-01-14:
As said everything is in readme and FAQ -  Happy printing

closed 2017-01-14 by luc-github