luc-github/Repetier-Firmware-4-Davinci #220: Davinci 1.0A w/ Repetier 0.92 not printing but takes some manual commands

ccpeters04, 2017-01-14:
I recently put Repetier firmware 0.92 on my XYZ DaVinci 1.0A and have been trying to get the printer to work.  I have installed Repetier Host software (tried 1.6.2 & 1.5.6) and get very odd results.  First, I cannot consistently get the printer to acknowledge manual commands sent from the PC (G28 works every time).  After slicing a model with Slic3r it looks like it is running a simulation (runs on the PC screen REALLY fast) but only the initial G28 and end home commands show any physical response from the printer.

Can you provide me with any suggestions as to what might be going on?

luc-github, 2017-01-14:
1- can you load and unload filament from printer without using host ? this is to see if you load fail safe settings properly
2 - did you select same baud rate from printer and host ?
3 - did check ping pong communication ? 
[image]

ccpeters04, 2017-01-14:
1) Yes, I have loaded filament from printer front panel
2) Yes, same baud rate
3) Ping pong checked.

Like I said, it works with some commands some of the time.  For example, I tried G28 x y z and it will do two of the three axes.

I have a hard time getting the PC manual control buttons to do anything a lot of the time, but manually entered text commands work quite a bit of the time.

luc-github, 2017-01-14:
G28 wihout arg should do all axis

what the serial log display when you send commands ?

ccpeters04, 2017-01-15:
A bare G28 command DOES home all three axes.

The commands all look correct in the log,  and with the echo on,  they are relayed back add they were sent.

I stepped through the slic3r generated gcode for a part, manually entering each line, and had some luck.   There were a couple of the start gcode that didn't react correctly.   There is an initial extruder retract (in my case,  a bed drop) that it didn't like. (G1 Z5 F5000)  I got it to take I'd I entered G1 Z5.0 F5000, but not consistently.   Subsequent move/extrude commands worked (i tried about 20 lines of gcode).

luc-github, 2017-01-15:
your Z speed acceleration is way too high so motor miss steps and at the end do not move 
 5000 is too high - check and change your slicer settings - 2000 is the max if I remember

ccpeters04, 2017-01-15:
So, I made some changes (lower speed settings) throughout repetier host and slic3r, but now I cannot get any g0/g1 commands (or manual control button actions) to make any of the axes move.  I tried setting the values back to shat they were as well, with no luck.  powered off the printer and powered it back on.  Unplugged USB and reconnected, no change.  Any ideas?

luc-github, 2017-01-15:
well I would say the issue is in your host settings definitly

the best is to remove repetier host and reinstall it to get clean default values

once host is ok you can check the slicer settings - slicer is not used for manual control

ccpeters04, 2017-01-15:
update...This is sort of what I was talking about earlier.  So, no single axis move works (ex. g1 z10 f1000), but if I fill in the other axis positions it works (ex. g1 x0 y0 z5 F1000).

luc-github, 2017-01-15:
you type or you use the manual tab ?

before each time you enter command manualy you need to enter G90 first

luc-github, 2017-01-15:
I cannot reproduce your problem - so definilty check your settings
[image]

ccpeters04, 2017-01-15:
I am re-installing rep host 1.6.2.  If this doesn't help anything I'll try the g90 before each command (which I didn't need to do the other day).  I have been able to use the manual control tab g-code text entry line and the buttons, but aside from the G28 command from the g-code line entry, I got very hit and miss reactions from the printer.

ccpeters04, 2017-01-15:
[image]
still no luck.  commands are echoed back, but no response.  Is there something I should check on the actual printer?

ccpeters04, 2017-01-15:
now this seems to be a consistent reaction. g28, OK, all axes home OK.  g90, OK. g21, OK.

ANY single axis input, via a g1 command, no response.  G1 command with all three axes positions specified, OK, all axes move to location.  After THIS command all single axis commands work 100% of the time as well as all of the manual control buttons.

luc-github, 2017-01-15:
g1 or G1 is seen as same command from fw so I do get why it act like this on your printer 

also I type same commands and everything works as expected digging issues it looks like this one : https://github.com/luc-github/Repetier-Firmware-0.92/issues/208

but I cannot reproduce

ccpeters04, 2017-01-15:
Added the M100 command to the custom start g-code in slic3r.  Movement restored!!Now I have one other issue.  The clean procedure (looked in the arduino code) sets the bed height to min +15mm.  In my case, that turns out to be 12.5mm, and that's where it goes, but it doesn't move to the correct  first layer position (stays at 12.5mm).  The second layer Z position move is correct, as are the following layer positions.  I even put in an extra Z position after clean, and a delay, but the first layer will not work.

ccpeters04, 2017-01-15:
After adding in a couple of lines to position all three axes after cleaning I was able to get complete prints.  Now I will just need to play with settings.  Thanks for the help.

luc-github, 2017-01-19:
ok so I close issue then

closed 2017-01-19 by luc-github