luc-github/Repetier-Firmware-4-Davinci #221: LCD support for volumetric extrusion?

JBKingdon, 2017-01-19:
Hi,
I'm just starting to think about using volumetric exfrusion, hoping that I can overcome some of Cura's limitations that way. If I understand correctly, to use volumetric extrusion I need to set the filament diameter on the printer before starting each print, and I can use M200 to do so. Which is fine if I'm printing from a host, but I typically print from SD and don't have a computer connected. Is there any LCD support for setting filament diameter for each extruder?

Regards,
James.

luc-github, 2017-01-19:
I did not check the latest repetier code but I doubt it is on it as original repetier FW rely a lot on host, I have introduced several input from lcd as I also print from SD only - this one is not part of them - but I do not see any issue to add more, what you describe seems just affect UI part (menu and translations) but can you explain more as I am not familiar with volumetric extrusion usage 
* this command : `M200 T<extruder> D<diameter> - Use volumetric extrusion. Set D0 or omit D to disable volumetric extr. Omit T for current extruder.`
need to be send before every launch print ? 
* does the values need to be stored in EEPROM or use default value that will be changed for each usage

JBKingdon, 2017-01-19:
I've only just started looking into volumetric extrusion, so I can't claim to be  familiar with the way it works in practice. Conceptually you'd need to set the filament diameter every time you loaded a different size filament for a given extruder. If you typically use the same filament most of the time you might not need to change the setting very often, so saving it to EEPROM could be a useful time-saver. On the other hand, if you have a lot of different filaments that you switch between frequently you'd need to be changing the values most times that you loaded filament. We do need to be able to set D independently for each extruder.

As an aside, I think the ideal case would be to add an rfid chip to each roll of filament that stores D for that roll, and have the printer read the value automatically, but being able to set the value on the LCD would be a good start :)

luc-github, 2017-01-19:
Hmmm setting a value on printer is not really a problem

the question is once this value is set, what do we do with this value ? :smile: 
what are the steps to use this value ?

* add another command to send the M200 for each extruder when selected ?
* ask to apply or not the M200 command using defined D value for each extruders when a SD Print is requested from panel ?
* how to handle the remote SD print ?-> do we need to add a custom command that apply M200 using defined D value for each extruders ?

The usage need to be defined based on all user cases - don't you think ?

JBKingdon, 2017-01-19:
I think as a first pass, it should behave exactly as if the printer received M200 from the host. Whatever the firmware currently does when it receives M200, we do that. Which precludes saving to eeprom as I don't think m200 is currently persistent. Once we have that working user feedback will soon tell us where it can be improved :)

luc-github, 2017-01-19:
Ok feel free to push a PR - I won't have time to work on this before a while, but I can help to review code

JBKingdon, 2017-01-19:
Oops, took me until now to realise I didn't post in the original Repetier repository. Sorry.

closed 2017-01-19 by JBKingdon