luc-github/Repetier-Firmware-4-Davinci #222: Davinci 2.0 wont flash correctly

rcryguy, 2017-01-26:
I have added the files to the needed directories and get no errors when i compile the firmware.
when i upload the firmware, the backlight on the screen turns on but the black bars are still there and the computer says that the device isnt working properly. The device will connect/unconnect many times during this period. tried clearing and re flashing many times

luc-github, 2017-01-26:
did you checked the FAQ ?

rcryguy, 2017-01-26:
Yes, couldn't find anything to fix it. Everything seems to be correct

luc-github, 2017-01-26:
let see
Where is your variant.cpp give me the full path
what is your OS ? 
what is your ide version and due module version from module manager ?

rcryguy, 2017-01-26:
I'll get the path of the varient a bit later. Win 10 ide 1.8.0 due 1.6.8

rcryguy, 2017-01-26:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants is the folder it is in

to clear it up, the firmware is loaded to the printer but the printer doesnt boot up and is no longer recognized on the pc

luc-github, 2017-01-26:
yes this location is wrong
as said in FAQ this is https://github.com/luc-github/Repetier-Firmware-0.92/issues/172
so read https://github.com/luc-github/Repetier-Firmware-0.92/issues/114 to find good location for variants.cpp there is a trick using preference to get hidden folder

rcryguy, 2017-01-26:
C:\Users\rcryg\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

opened it up and it was in here as well

rcryguy, 2017-01-26:
copied the repetier files there just in case but still problem, excpet the pc sees it as arduino DUE now after flash

luc-github, 2017-01-26:
yes this one is correct this is the place variant.cpp need to be modified

rcryguy, 2017-01-26:
im going to reload arduino real quick, but still no luck

luc-github, 2017-01-26:
did you verified you modifed the file and not copy to another location ?

rcryguy, 2017-01-26:
Can you explain a bit? i replaced the variants.cpp with the one include in the repetier download

luc-github, 2017-01-26:
Some people do a paste without checking the file is well replaced

One thing : you need to replace files with the ide closed, if ide is open changes may not be taken in account

luc-github, 2017-01-26:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/213#issuecomment-269435300

rcryguy, 2017-01-26:
did that exact thing, i get this while compiling but it uploads anyway
[image]

luc-github, 2017-01-26:
yes this is a warning - it is not an issue

rcryguy, 2017-01-26:
the rpinter still doesnt boot up on repetier, the backlight of the screen turns out but thats it. the black bars are still on the screen andwont respond to any commands in repetier host

luc-github, 2017-01-26:
as originaly you did not do the right changes in right place you  may mess up something - I suggest you to uninstall the due module - uninstall arduino - wipe the board with jumper and restart process from scratch

rcryguy, 2017-01-26:
just unistalled and reinstalled ide and due as long as a fresh download of repetier... still nothing

luc-github, 2017-01-26:
and replaced variants.cpp of : C:\Users\rcryg\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\variants.cpp ?

because black bars are wrong variants.cpp 100%

rcryguy, 2017-01-26:
is variants.cpp supposed to be int the sketch to uload to the printer?

luc-github, 2017-01-26:
sorry what process are you following to replace variants.cpp ? you confuse me

rcryguy, 2017-01-26:
i copy the arduino15 folder to C:\Users\rcryg\AppData\Local and click replace on the popup warning me of the duplicate names

rcryguy, 2017-01-26:
the folder i copy at first is from the firmware downoad

luc-github, 2017-01-26:
and do you have this comment line 326 ?
  **// 125**
of 
C:\Users\rcryg\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x\variants.cpp

Cannonized, 2017-01-26:
What version of arduino I had to modify a find line in the software using 1.6.5r2 I posted the fix as a comment in one of the posts I'll see if I can find it.

rcryguy, 2017-01-26:
line 326 is highlighted
[image]

rcryguy, 2017-01-26:
ide version 1.8.0 due version 1.6.8

luc-github, 2017-01-26:
do you restart printer after flash is done ?

rcryguy, 2017-01-26:
yes

luc-github, 2017-01-26:
well I am dry - but my line 326 is not the one you highlighed
[image]

rcryguy, 2017-01-26:
so the black lines indicate that the variants.cpp is not correct?

rcryguy, 2017-01-26:
just recopied so my line 326 lines up with yours

rcryguy, 2017-01-26:
[image] this is the output after upload

rcryguy, 2017-01-27:
[image]
im trying arduino 1.5.8 now, and this is the error i get... maybe it will help find the problem? 
Any additional help is greatly appreciated because right now i have an expensive brick

rcryguy, 2017-01-27:
Major breakthrough. I was able to take a bin file of the old firmware and upload it so I no longer have a brick, but can you create a bin file for this firmware?

luc-github, 2017-01-27:
1.5.8 ide is not supported by current git version firmware, https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.3.0 should work with 1.5.8

Yes if bin work - it confirm you do not follow the readme in someway and mess up with the code

luc-github, 2017-02-16:
As no update I close issue

closed 2017-02-16 by luc-github