luc-github/Repetier-Firmware-4-Davinci #224: Please include 1.0 Pro support

DavinciPro1, 2017-01-27:
That would be nice, I want to have full control over S3D over my maker aswell ;)

luc-github, 2017-01-27:
Duplicate of https://github.com/luc-github/Repetier-Firmware-0.92/issues/142 https://github.com/luc-github/Repetier-Firmware-0.92/issues/164

as mentioned in FAQ https://github.com/luc-github/Repetier-Firmware-0.92/issues/106

luc-github, 2017-01-29:
if no update I close issue

closed 2017-01-29 by luc-github

Quoti86, 2017-08-07:
:-( i realized that octoprint/octopi only works with repetier firmware on davinci printers. But my Problem is the same: i have a Davinci 1.o "PRO", so i cant use repetier and octoprint too :-( this makes me very sad, becaue i share the printer with a good friend, and i would love to use octopi...is there no chance to get support?

DeeCee2501, 2018-12-22:
Is somebody working on a FW for the Da Vinci Pro? I would like to flash my Da Vinci Pro 3in1 to get a Titan Extruder working on it (by changing steps per mm value and extruder direction inverse) and I wouldn't like to change the whole stock mainboard for it. 
My first idea was to run the stock firmware through a disassembler and trying to modify the code, but I couldn't get the stock firmware for it yet... :-(

luc-github, 2018-12-22:
Not me  - also if the board is same as JR, as it is same MCU (SAM4E8E), Davinci use a separate mcu ( M0) for temperature control making the open FW porting even more difficult.
Today only reprap FW support SAM4E8E but flashing the M0 FW to control the temperatures is the stopping issue as pinout is known on JR so should be possible on Pro