luc-github/Repetier-Firmware-4-Davinci #225: Fan for PLA

Hannes54, 2017-01-28:
Hello luc
My printer works with the new firmware faultlessly.
I have a 2.0A with 2 fans.
Now I would like to operate a fan for PLA.
Would it be possible to use the output of the Mightyboard to switch the lighting? I can also use the light manually.

Best regards and thank you for your work

Hannes
P.S. Sorry, I use Google translator.

luc-github, 2017-01-29:
why not use https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L26

Hannes54, 2017-01-29:
Hello Luc,
Thank you for the fast answer.
If I understand this correctly then I use one of the two extruder fans for cooling. But this has no function to cool the extruder. Is that correct? Which extruder is the "main extruder"?
I would like no change both extruder fans because I then expect problems.

best regards
Hannes

luc-github, 2017-01-29:
you just connect right fan to left fan so both work in same time which is fine 
and use right fan connector to connect you cooler fan

Hannes54, 2017-01-29:
Hello Luc,
Thank you, that sounds good. I'll try that out.

The old problems with the temperature of my printtable are solved. I have built a new printtable. Now a 100k thermistor is used. With a new temperature table it works very well.

Without your firmware I would have sold the printer certainly already!

Thank you!

greeting
Hannes

luc-github, 2017-01-29:
great, I close issue then
 happy printing

closed 2017-01-29 by luc-github

lern01, 2021-12-29:
> 您只需将右风扇连接到左风扇，这样两者可以同时工作，这很好， 并使用右风扇连接器连接您的冷却器风扇

How to control the two fans separately?

lern01, 2021-12-29:
> why not use https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L26

How to set up one fan with temperature control and the other fan with code control

luc-github, 2021-12-29:
as mentioned :
```
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 //Setting this to 1 will repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need to add a permanent fan with power supply to cool extruder
```
set to 1 allows to use current control pin for extruder fan to be used for fanduck - but you need to change wiring  of extruder fan to powered it - no extra pin available so far, so the fan will always work, one solution  is that you may use the led pin to keep control of this fan and let led light always on

lern01, 2021-12-29:
Thank you very much. It's been taken care of

lern01, 2021-12-30:
After the left and right fans are connected together, the output voltage is only 3.1V. How to solve this problem?

luc-github, 2021-12-30:
the fans are 5v there is some 5v on main board if I remember well

lern01, 2021-12-31:
The working voltage of the fan is 5V, after connecting two fans, the voltage drops to 3.1V, while connecting one is normal, I don't know what the problem is.  Measured the triode C pole, connected with a fan, when the triode on, C pole 0V, connected with two is about 1.8V.  The fan voltage is only 5-1.8=3.2.

lern01, 2021-12-31:
My model is Da Vinci 2.0 A Duo.

lern01, 2021-12-31:
I checked the data and found that the working current of the triode is 600mA and the circuit is simple. The CUP output is connected with 1K resistance to the B pole of the triode, the C pole is connected to the negative pole of the fan, and the E pole is grounded. Remove the two fans and connect them to the adjustable power supply. The current is about 300mA. Is the CPU output a PWM signal?

lern01, 2021-12-31:
Both fan interfaces on the motherboard have been tried, and the result is the same, indicating that the circuit is ok  
I'm a little confused. Even with PWM output, if I set it to 100% and connect it to two parallel fans, the result is the same regardless of interface 1 or interface 2.

lern01, 2021-12-31:
Luc,
hello!
Can you communicate in Chinese? My English is really poor. thanks!

luc-github, 2021-12-31:
Sorry my chinese is basic just daily life speaking and very liw for reading / writing.

The pin output is 3.3v  on Davinci it is normal

lern01, 2021-12-31:
I said that the voltage between the two ends of the parallel fan is only about 3.1V, normally it should be about 5V 。 
I'm really sorry for bothering you so much！

lern01, 2021-12-31:
May I ask how to adjust the PWM of the extrusion head fan in the firmware?  I suspect there is no 100% output in the firmware.