luc-github/Repetier-Firmware-4-Davinci #228: still black screen,

jepolqwerty, 2017-02-27:
Hi, 

    I followed all the steps correctly, and watch the tutorials, but still i get black bars in screen please help

luc-github, 2017-02-27:
what is your printer?
did you read the FAQ?

jepolqwerty, 2017-02-27:
may printer is da vinci 1.0A, yup i read the FAQ, i cannot find the FAQ#171
<https://github.com/luc-github/Repetier-Firmware-0.92/issues/201>, but
there is FAQ #172, but i followed that also

jepolqwerty, 2017-02-27:
after the sending uploading in the printer, it disconnect and then it
continuously says that unrecognized USB

jepolqwerty, 2017-02-27:
[image: Inline image 1]

after the uploading, it continuously says that Usb device not recognized.

Please help

luc-github, 2017-02-27:
good catch typo issue
your issue is not correct update of variant.cpp

jepolqwerty, 2017-02-27:
how do i fix it?

please instruct me.

jepolqwerty, 2017-02-27:
im using arduino IDE 1.8.1

and arduino SAM board 1.6.11

luc-github, 2017-02-27:
follow the readme exactly, and check Faq for details

using sam 1.6.11 is a proof you do not follow the readme, I never tested sam 1.6.11 so issue could come from there

jepolqwerty, 2017-02-27:
i reset again the printer, change the sam to 1.6.8 copy the hardware from
AdditionalArduinoFiles, i still cant make it work

luc-github, 2017-02-27:
what is the path of the variant.cpp you update?

luc-github, 2017-02-27:
you solved your issue as mentioned here[#229] (https://github.com/luc-github/Repetier-Firmware-0.92/issues/229) ?
How did you do ?

jepolqwerty, 2017-02-27:
I read the comments in other tutorials and they have the same problem as I
encounter. So what they did is to downgrade the IDE and install the older
version of repetier. I did it also and it works

luc-github, 2017-02-27:
so which version of IDE did you used ?

jepolqwerty, 2017-02-27:
I used the 1.5.8

luc-github, 2017-02-27:
ok so you did not used the git version of fw but the 0.93.2 based

jepolqwerty, 2017-02-27:
But im still playing the printer now. I think even though I made it. I
think the firmware is mess up. The light is not working in I cant made the
z axis to move up. Cant connect to the software. And I cant even load
filament.

jepolqwerty, 2017-02-27:
There is already .93.2? Im trying to work before is the version 92.10

luc-github, 2017-02-27:
I guess EEPROM is not correctly set - did you do the load fail safe seetings  ?
what fw version of FW are you using ?

luc-github, 2017-02-27:
Sorry I mean 0.92.3 
0.92.10 do not compile with 1.5.8 so I do not get

jepolqwerty, 2017-02-27:
I cant Remember, ill check it tomorrow. But it was version made late 2015

luc-github, 2017-02-27:
if it was made in 2015 it is outdated I suggest you to use latest git version

closed 2017-02-27 by luc-github

jepolqwerty, 2017-02-28:
I tried the 1.6.9 version and up and still the same, i cant made it work. i
dont know whats the problem. i followed all the instruction well, specially
need to copy the "arduino15 folder" inside the "additional folder" but
still i cant manage to install it. im starting to loose hope

luc-github, 2017-02-28:
I have added more check on https://github.com/luc-github/Repetier-Firmware-0.92/issues/114

jepolqwerty, 2017-03-01:
[image: Inline image 1]if your referring this, i cant find that version

luc-github, 2017-03-01:
I do not see what image you linked

jepolqwerty, 2017-03-01:
SAM version 1.6.2 BETA

luc-github, 2017-03-01:
This is not the version I mention in read me , it is old and file location is different than latest version

jepolqwerty, 2017-03-01:
that is in the picture in #114

jepolqwerty, 2017-03-01:
i cant make the latest version worked, before it finished it disconnect.
the older version i mange to send it to the printer, but the old version is
messed up. do you have any older version? newer than
https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.3

jepolqwerty, 2017-03-01:
but older version of this,
https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.3.0

luc-github, 2017-03-01:
the latest one is this one:
https://github.com/luc-github/Repetier-Firmware-0.92/archive/master.zip

jepolqwerty, 2017-03-01:
my printer now is working, thanks in reviewing all the video tutotials

i have another problem now, i cant used simplify 3d in the printer

luc-github, 2017-03-01:
Good - may I know what was the issue ? 
what step was different  with read me or FAQ ?

jepolqwerty, 2017-03-01:
i still cant manage to make latest version work. i used the older version,
the reason why the firmware send to my printer is messed up, is because i
did not select the J37 port,

i manage to connect the printer to repetier host but i cant send gcode to
the printer, do you have settings?  can you share it to me?

luc-github, 2017-03-01:
Did you used the settings I shared ? https://github.com/luc-github/Repetier-Firmware-0.92/issues/229#issuecomment-282775061

what you mean by send GCODE, when you press home in manual tab is it not working ?

jepolqwerty, 2017-03-01:
> Did you used the settings I shared ? #229 (comment)

I cant control the printer using the repitier host and when I hit print
nothing is happen to the printer its not even heatingqup

luc-github, 2017-03-01:
Did you load failsafe settings from printer like wrote in readme?
Or send M502 M500 from repetier host?