luc-github/Repetier-Firmware-4-Davinci #229: cant connect to slicing software

jepolqwerty, 2017-02-27:
Hi,
   I managed to flash the firmware its just that now i cant connect to the printer, in simplify3d it can find, but cant connect, in xyz ware, totally cant find

luc-github, 2017-02-27:
simplify3d can connect repetier - check baud rate - uncheck wait for start and be sure hardware flow control is  checked

 xyz ware is not working of course - it is no more xyz firmware - but all hosts that can connect with repetier/marlin FW can connect

jepolqwerty, 2017-02-27:
Even in repetier host. I cant connect. The printer keeps rejecting the
connection. How about in the printer do I need to configure anything

luc-github, 2017-02-27:
here my settings - I close issue as not fw related
[image]

closed 2017-02-27 by luc-github

jepolqwerty, 2017-02-28:
Thank you, but even though i manage to connect the printer. i cant use it,
i need to reflash the printer