luc-github/Repetier-Firmware-4-Davinci #230: Built server

gamecompiler, 2017-03-02:
Hi luc (Sorry for my bad english in advance)
im may allowed to built the source code for different printers out of your firmware and publish the bin files with an console based installer? (May be you dont like it if i do that, i would like to also include a firmware ready to make use of a e3d hotend)

In the past id have many problems compiling your firmware (especially windoof and mac).
I thought it may help other users if im compiling them the firmware and bundle it (optionally) with an console installer (just a shell script using bossac - my favorite tool)

Thanks in advance,
gamecompiler

luc-github, 2017-03-02:
well it is in my todo list to update travis to compile and save all binaries - I do it already for another project but on private server - I just did not had time to read carefully the travis doc for generated binary transfert
 
As fixes come time to time you may have hard time to maintain the binaries but feel free until the travis-ci is update
for sure having an flash tool would be great , 99.99% of flashing issue is people just do not follow read me and check outdated video / tutorial and messup and panic and read even less instruction

because the compilation is only take sources / change 2 files and push flash button but this generate big frustration  as people do not follow instructions....

luc-github, 2017-04-03:
Hi any update on this ?

gamecompiler, 2017-04-03:
Yes i will post my progress on Saturday. Im writing my advanced math 
exam friday.

"Electrical Engineering" is a very hard study.

Im sorry, but I have to prioritize my math exam.

The built server is ready, its just an jenkins CLI uploading the bin 
files to my http server and lists those.

In future i would like to have an interactive gui for the firmware.

---------------------------------------------
Der perfekte Zeitpunkt seinen Gasversorger zu wechseln:
http://www.anpa.de/index.php?action=gasvergleich

luc-github, 2017-04-03:
No problem - No hurry  -I was just looking for update to see if still live

Good luck for your exam

gamecompiler, 2017-04-03:
Thank you very much. I think id learned much.

Could you please give me your email address? I would like to give you an link that triggers an build everytime you click on it. I think you are the person that perfectly know when its time to trigger it.

gamecompiler, 2017-04-03:
You can take a look at the build if you want. Under the directory build are the binaries.
Its sorted as following DAVINCI->MODEL
For example for Davinci 0,Model 0 the binaries are located in build/0/0

gamecompiler, 2017-04-03:
http://expert-craft.com:8087/

luc-github, 2017-04-04:
Hi to communicate such information it is better not public, as it is your private server and can easily overflooded 
I suggest you to use gitter and send one to one message - so only me will see it and it is archived also
https://gitter.im/luc-github/Repetier-Firmware-0.92

luc-github, 2017-04-10:
Hi I have updated travis to autobuild binaries -  got some times this week end and this morning to fine tune - I did not tested the flash yet but as it compiling it should be Ok - I have added the USBCore patch also. 
https://github.com/luc-github/Repetier-Firmware-0.92/releases/tag/v0.92.10rc2d
The binaries won't be generated at each commit but only for each tag

Let me know when your flash tool is ready I will try it

gamecompiler, 2017-04-10:
Hey nice ^^

On weekend im working on the flash utility. Its currently working, but ive got some problems while listing the serial ports. Its just a simple python script.

luc-github, 2017-04-14:
Ok let me know if you need some help or test monkey

I close issue as auto build is now ready

closed 2017-04-14 by luc-github