luc-github/Repetier-Firmware-4-Davinci #231: DaVinci 1.0 Freezing and Restarting Every 24 Seconds

NoahSkr, 2017-03-04:
I just got a DaVinci 1.0 and installed Repetier Firmware 1.0 on it, but the printer will only stay on for 24 seconds. I tried disabling watchdog, but then the printer just stayed on and froze after 24 seconds. What is going on here? I would really like to get this printer working.

luc-github, 2017-03-05:
what is your printer model ?
what ide version do you use ?
what due module version do you use ?
Did you do any change in configuration.h ?
What FW version do you use ? 
Did you do any hardware modification ?

NoahSkr, 2017-03-05:
-Printer Model: Davinci 1.0
-Arduino IDE: 1.8.0
-Due Version: 1.6.8
-Changes to configuration.h: none
-Firmware version: I tried .92 (the version that downloads when the green download button is clicked.)
-Hardware Mods: None
-Screen: Fully functional and works fine until 24 seconds, then it freezes and restarts due to the watchdog.
-I don't know how to find a serial output. If that is found in the serial monitor, it says "wait" twice, but it just connects and then disconnects.

luc-github, 2017-03-05:
Did you update usbcore.cpp ?
Does it happen if printer is not connected by USB ?
Looks like init is OK so I only can imagine is a dead loop but I do not see why
Did you try wipe with jumper again and flash again ?

luc-github, 2017-03-05:
Can you share  the version displayed when printer is starting ?

NoahSkr, 2017-03-05:
Using Arduino 1.5.8 to upload fixed the problem to upload to my DaVinci 1.0. I'm not sure why, but it worked. Using 1.8.0 worked only on the 1.0a for me. It doesn't restart any more, and I finally got a decent print. Thank you for your time and help.

luc-github, 2017-03-05:
If you are able to  compile with 1.5.8 it means you do not use latest git version, be aware auto level is not working well then, so better not use it

Happy printing

closed 2017-03-08 by luc-github