luc-github/Repetier-Firmware-4-Davinci #233: Manual bed leveling with change in extruder height.

ngen33r, 2017-03-06:
Davinci 1.0 JP1 version
Latest firmware .92.10M
Prints great (fan/thermistor issue solved by using shielded wire, not a firmware issue)

I changed over to an E3D hot end and lost 23.6mm in Z. Z min pos is set to -23.60 in eeprom so after homing I can print. When manual bed leveling on machine the head crashes into the build platform. Can we add an option to move safe Z (Z min pos * - 1 = 23.60) and maybe Y 100 before starting leveling. This will allow for any extruder modification and allow the leveling on the machine to work as intended without any risk of collision.

luc-github, 2017-03-06:
well again bed leveling is supposed to be done at Zmin = 0
How you can rely on any value of Zmin the first time you do bed leveling 
1 - you may have a wrong value actually it is 100% sure you have a wrong value if you change extruder
so what will prevent the crash ?
2 -what behavior do you expect if you do a Z homing with extruder in middle of the bed ? 
3 - Zmin is supposed to fine tune the zmin with small values not so big value as 23mm

That is why modding extruder need Z home sensor modding to have Zmin = 0 not hitting bed  by moving sensor down or by upping flag

finally one comment of my (long) experience in software/hardware development : workaround a hardware design problem by software is bad idea - just see point 2 => big crash

ngen33r, 2017-03-09:
I agree with points 1 and 2 but if the offset is measured then you should have a way to tell the machine what you want it to do. I have a $250,000 CNC machine in my shop that homes at -100cm and it can always find zero. My printer homes at -23.6mm. I know where home and zero are. I just need the software to know where home and zero are.

luc-github, 2017-03-10:
home is when you reach sensor - can be z min or z max 
>I have a $250,000 CNC machine in my shop that homes at -100cm 

 you mean when you home your bed is 1 meter higher than the extruder ? is that ? what kid of printer do that I am curious ? it is not a delta which home at +100cm instead ?  (using Zmax)

Adding a menu would be confusing - but adding a define at compilation can be a solution to use, but it won't prevent the crash on bed if homing Z when extruder is on middle of the bed and if at home extruder position is lower than the bed

ngen33r, 2017-03-10:
My example is not good. The machine I reference has feedback of position. Our 3D printers do not have feedback.

I now understand why you recommend this not to be done. Homing Z the machine does not know that the switch is above the zero point and will crash every time. 

A workaround for this will be to make the machine always home X and Y first when a home Z command is sent, but would probably be better to just move the sensor and be done with it.

I know understand why it is better to move the sensor to machine zero.

closed 2017-03-10 by ngen33r