luc-github/Repetier-Firmware-4-Davinci #234: Davinci 2.0 stopped working??

bbozzi12, 2017-03-10:
i was in the middle of printing and the front display just turned off. I thought it was a software issue so i tried re flashing the printer and now none of the lights are turning on or the front display. What do I do?

bbozzi12, 2017-03-10:
i just tried uploading repetier onto the printer and nothing happened, the red light on the board is still on but the front display and lights are still off.

luc-github, 2017-03-10:
looks more hardware issue to me, like a short or hardware failure

did you wipe with jumper? what does screen display ?

luc-github, 2017-03-10:
looks more hardware issue to me, like a short or hardware failure

did you wipe with jumper before reflashing?

bbozzi12, 2017-03-10:
i wiped with the jumper and everything turned off

bbozzi12, 2017-03-10:
originally when the front display first went off, i stopped the print and it wouldnt let me control the heaters anymore, so i turned the printer off and back on and it still wouldnt let me control the heaters. So i wiped the printer and all the lights went off.

luc-github, 2017-03-10:
As wrote before looks like an hardware problem, if reflashing did not helped it is confirmed

bbozzi12, 2017-03-10:
do you have any idea what hardware it could be

luc-github, 2017-03-10:
nope sorry, looks like you can detect usb and flash so should not be MCU issue, I would guess for a short, unplug all and retry flashing ang check if serial output send something
if ok then connect back screen to see if display is ok, then others connectors one by one

bbozzi12, 2017-03-10:
thanks anyways, this is my 3rd davinci, they always give me issues

luc-github, 2017-03-10:
davinci is very fragile, wires are easy to break, especialy new generation

I cross my fingers, my 2.0 has only one loose connector issue since I got it

luc-github, 2017-03-10:
I close issue as Hardware issue

closed 2017-03-10 by luc-github