luc-github/Repetier-Firmware-4-Davinci #235: Compilation error

Ethydee, 2017-03-18:
trying to flash da vinci 2.0a but i keep getting compilation error
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Error Code:

Arduino: 1.8.0 (Windows 10), Board: "Arduino Due (Native USB Port)"

core\variant.cpp.o: In function `USART3_Handler':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:314: multiple definition of `UART_Handler'

sketch\variant.cpp.o:sketch/variant.cpp:357: first defined here

core\variant.cpp.o: In function `_GLOBAL__sub_I_g_APinDescription':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:341: multiple definition of `USART0_Handler'

sketch\variant.cpp.o:sketch/variant.cpp:384: first defined here

core\variant.cpp.o: In function `USART1_Handler':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:346: multiple definition of `USART1_Handler'

sketch\variant.cpp.o:sketch/variant.cpp:389: first defined here

core\variant.cpp.o: In function `USART3_Handler':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:351: multiple definition of `USART3_Handler'

sketch\variant.cpp.o:sketch/variant.cpp:394: first defined here

core\variant.cpp.o: In function `init':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:373: multiple definition of `init'

sketch\variant.cpp.o:sketch/variant.cpp:416: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `Serial3'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `Serial2'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `Serial1'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `tx_buffer4'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `tx_buffer3'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `tx_buffer2'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `rx_buffer4'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `rx_buffer3'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `rx_buffer2'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `Serial'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `tx_buffer1'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `rx_buffer1'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o: In function `serialEvent3()':

C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x/variant.cpp:309: multiple definition of `g_pinStatus'

sketch\variant.cpp.o:sketch/variant.cpp:352: first defined here

core\variant.cpp.o:(.rodata.g_APinDescription+0x0): multiple definition of `g_APinDescription'

sketch\variant.cpp.o:(.rodata.g_APinDescription+0x0): first defined here

c:/users/ethydee/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `g_APinDescription' changed from 3668 in sketch\variant.cpp.o to 2604 in core\variant.cpp.o

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Native USB Port).

luc-github, 2017-03-18:
Message is pretty obvious:  there are 2 variant.cpp when should have one 
you did not replaced the existing one, but added a second one in different place so there is a problem of course.
I suggest you to remove the due module by board manager a after the directory C:\Users\ethydee\AppData\Local\Arduino15\packages\arduino\hardware\sam

then reinstall due 1.6.8 module (and not another version)

Once done go to directory C:\Users\user\ethydee\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x
to **REPLACE** the variant.cpp with the one from github fw

Ethydee, 2017-03-18:
OK that worked thanks

closed 2017-03-18 by Ethydee

luc-github, 2017-03-18:
Good :smile: Happy printing