luc-github/Repetier-Firmware-4-Davinci #236: Da Vinci Duo Chimera upgrade???

jamskate, 2017-03-27:
Hello,
I have created an upgrade to install the E3D Chimera in the Duo 2.0.   
http://www.thingiverse.com/thing:2202297

Can someone tell me how I get rid of the commands that send the carriage to the left side of the print bed when loading the left extruder?  My new design is small enough that both heads fit over the right drip tray and I want to remove the left tray to install a larger print bed.  

Thanks
Jamskate
"If it ain't broke don't fix it.   UPGRADE IT!!!!"

luc-github, 2017-03-27:
I think these 2 lines:
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/ui.cpp#L4865

https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/ui.cpp#L4877

but have others command that may need change like clean extruder function

jamskate, 2017-03-27:
Thanks,
I'll give that a try tonite.  If that works I'll need to eliminate all commands that utilize the left drip tray.

jamskate, 2017-03-28:
That did the trick.  Now it stay over the right drip tray for left or right load/unload.  

Now, is there any chance you can help me take out the part of the cleaning process where it wipes on the left drip tray?

Thanks
 Jamskate
 "If it ain't broke don't fix it. UPGRADE IT!!!!"

luc-github, 2017-03-28:
the clean noozle function is here :
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Printer.cpp#L208-L316

jamskate, 2017-03-28:
Ok, so i'm not the best at coding am I correct in assuming I just need to delete lines 280-303?

Thanks again

luc-github, 2017-03-28:
All depend what you expect the extruder do for cleaning the 2 nozzles
extT == 1 means if extruder 2
extT == 0 means if extruder 1 
IMHO it is not just remove code but limit the movement by not using xLength (which move to left dropbox)
you have to create your own path for clean extruders for the chimera I think

jamskate, 2017-03-28:
Actually the two print heads are so close together I believe if I just relocate the wipe panel in the drip box slightly to the left it would clean both heads at the same time.   If that's the case just the first portion that cleans the right extruder would do the job.

jamskate, 2017-03-29:
Thanks for the pointers.  I managed to rewrite the code to wipe extruder 1 three times across the cleaning tab then move left 18mm and wipe extruder 2 three times then move to the front of the printer.  Admittedly I still don't entirely understand the code but thru trial and error managed to piece together the commands I needed.  I couldn't have done it without you pointing me to all the code.

Many Thanks
Jamskate
"If it ain't broke don't fix it. UPGRADE IT!!!!"

luc-github, 2017-04-03:
Hi feel free to add a wiki page on how to setup your solution - wiki is open 
if you can list all changes in wiki I may be able to add a flag for chimera setup to apply your changes

jamskate, 2017-04-03:
I've never used wiki but I'll look into it.   Guess I'll have to step up my learning curve.  Lol

Having a Chimera setup option sounds great.

luc-github, 2017-04-04:
No worry using wiki is like using notepad and all is archived so any revision can be recovered

jamskate, 2017-04-04:
Can you post a link to the wiki you use?  So far the ones I'm finding the links to add a wiki are broke or they have a monthly charge..

luc-github, 2017-04-04:
it is the wiki of this github, on top you have a button `wiki` https://github.com/luc-github/Repetier-Firmware-0.92/wiki
just push add page button to add a page

very simple

jamskate, 2017-04-04:
Ok, I'm totally out of my element here..  Am I making my own repository or adding a project under yours?  I don't want to be difficult but I don't understand how this site works and its starting to frustrate me..

luc-github, 2017-04-04:
and no worry, nothing can be broken ^_^

jamskate, 2017-04-04:
Well I did something!!!!  I posted the link to Thingiverse and was going to add my zip file of the Repetier folder but couldn't figure out how to add it.

jamskate, 2017-04-04:
Sorry I kinda just chopped up your code to make it work for me, I wish I was good enough at it to do like your thinking and just add an option 5 for Chimera.  Sad part is that as a kid I would have been considered a computer geek and learned to program in basic on the old TRS80's and Apple 2c's(giving away my age now)  guess I should have stuck with it..

Let me know if you need me to do anything else.

luc-github, 2017-04-04:
haha I learned on ZX81 and got an apple 2c too, so no worry 
Will check and comment later, thanks for your help

jamskate, 2017-04-04:
Btw I do have one question.    Every time I try switching from option 5 to option 8 for the thermistor option I get null readings on my display.   The only way I get readings is sticking with option 5.    According to E3D's directions I should be using 8.    Am I missing something?

luc-github, 2017-04-04:
may be https://github.com/luc-github/Repetier-Firmware-0.92/issues/104

jamskate, 2017-04-05:
Ok, so I'm thinking since most of us DaVinci owners are switching to E3D hotends a custom table needs to get made for the Semitec 104Gt2.   I know I get readings with them but I'm curious to know how accurate.    Can you point me to the custom table for option 5 and I'll try and figure it out.

luc-github, 2017-04-05:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/207

jamskate, 2017-04-05:
Ok,   I'll look at my config tonite.    That setting change we need to be made for both extruders in the Chimera build as well.

luc-github, 2017-04-05:
both use same table
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L734

jamskate, 2017-04-09:
So do I need to do anything besides change lines 365 & 507 to read as follows?
#define EXT0_TEMPSENSOR_TYPE 7
I just tried it that way and temps defaulted to zero and when I tried to heat it gave me the thermocouple disconnect error.

luc-github, 2017-04-10:
I cannot try this table as it is for new generation
check https://github.com/luc-github/Repetier-Firmware-0.92/issues/108 to verify the temperature reading

luc-github, 2017-04-22:
I change title for better tracking

jamskate, 2017-05-04:
Luc,
Currently startup screen displays the printer name and the Repetier version using the first and third display lines.  Is there a way to use the 2nd and 4th display lines?  I'd like to make my startup screen read as follows.

Da Vinci 1.0
   E3D V6
Repetier0.92.10M
NAK 3D Designs

Thanks

luc-github, 2017-05-04:
to change current boot screen you need to change here :
https://github.com/luc-github/Repetier-Firmware-0.92/blob/7f28c5eded5f35210429f42aaba72ec0e9796036/src/ArduinoDUE/Repetier/ui.cpp#L1227-L1231

jamskate, 2017-05-04:
Took a bit of trial and error but I got it to display what I wanted.  Thank you..

luc-github, 2017-05-04:
good! have fun ^_^

jamskate, 2017-06-01:
Hi Luc,
New problem...  I've finally finished my custom 200mm x 200mm printbed but have an issue.
On the Configuration.h tab I have the following setting to avoid my corner clamp.
Line 1051   #define ENDSTOP_Y_BACK_ON_HOME 7.
No matter what setting I adjust I can't get the print to go behind that 7mm offset to avoid the clamp when printing.

Am I missing something or is there another way to get the hotend carriage to move forwardx to avoid the clamps when homing and performing the clean cycle without excluding those 7mm when printing?

Hope this makes sense..

luc-github, 2017-06-02:
this setting let home Y then move extruder front to 7 mm, and this position is the maximum position, this avoid when moving X to hit the plate on the back.
you must load fail safe after changing this setting to recalculate the Y lenght accordingly or manually set new value of y lenght

you cannot go behind this offset unless you change y min/Y home position to 7 but doing this won't prevent to hit the plate in many case like homing when your extruder is in middle of x and at minimum Y position

IMHO it is better to loose some surface to avoid any hit

and if you ask : no there is no reliable way to avoid this plates and print using maximum surface, it is design issue of the bed or a limitation depending of point of view

closed 2020-12-13 by luc-github