luc-github/Repetier-Firmware-4-Davinci #238: Da Vinci 2.0 Duo Left Extruder Only

eriemaker, 2017-04-12:
Hi,
I am trying to figure out how to run my left extruder only in the repetier settings.  I did a swap where I put an E3D hotend in only on the left side, so the right side has no extruder.  I was getting a heater decouple error which I troubleshot and it is occurring because the right extruder is not there.  I found issue #94 , where Luc stated this: "ok yes so it is corner case, so your solution to set NUM_EXTRUDER to 1 will solve the issue but may bring some mess elsewhere".  

I thought that would fix my issue, but then I realized that when I did that it applies everything to the right extruder.  Is there a simple fix that I can do in the config file that would swap everything to tell it that I only have a left extruder?  One of my thoughts was to just connect the wires from the extruder, hotend, etc. from the right to the left, but then I will need to some sort of x offset as it will think its printing from right side?  Or I can pull apart the printer again and put the extruder in the right side, though I would like to avoid that.  

For what its worth, I did disable the decouple alert, and changed my simplify settings to print from left extruder, but when it warmed up I got an alert to load filament in the right extruder.

luc-github, 2017-04-12:
if you do not use preheat functions / clean function from printer menu you should not have any issue
using only GCODE (SD print or Host print) should not use the right extruder unless you set it

Left extruder is T0 and also the default one if none is defined - Right extruder is T1 - if your GCode do not use T1, there no reason to get any issue about right extruder not powered: see my S3D settings:
[image]

That said, decouple issue may also be related to the E3D hotend, especially if you do not use the stock thermistor but the E3D and did not changed the thermistor table or/and did not do the modification of resistor as explained in  FAQ

eriemaker, 2017-04-12:
Luc,
Thanks for the response.  I have my simplify setup the same exact way you do above, but after I disabled decouple and tried to print, it heated the bed and extruder, but then I got an error that I need to load the right side filament, what would cause that?  It seems as though it is still looking for the right side? 

Oh and I cannot use the manual bed leveling either because it wont start as it keeps looking for the right extruder to heat up, that is why I was going to switch to single extruder in the configh. 

Thanks,
Henry

luc-github, 2017-04-12:
it means it heat T1 if any command using T1 it check sensor - did you checked your generated GCODE ?
you can disable sensor also in printer menu

eriemaker, 2017-04-13:
Luc,
I just checked my gcode and there is no reference of T1 in the code.  I am going to try and disable the senor and I will report back in.  Thanks again for the help!

eriemaker, 2017-04-13:
I just ran it again, and I am getting "Extruder 1 has no filament! Push Ok to go to change filament"

eriemaker, 2017-04-13:
Ok so I disabled the filament sensor and it started printing, but it never raised the bed up to printing height.  I was waiting for it to raise the bed so I could shut it off and level the bed but it just started printing 1.5" or so above the bed, I am assuming that is in my gcode, but I never had that issue before?

luc-github, 2017-04-13:
what change did you do in configuration.h
it may see printer as davinci 1, in that case T0 is right extruder

eriemaker, 2017-04-13:
#define DAVINCI 3
#define MODEL  0
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0

eriemaker, 2017-04-13:
Luc,
Ok so I checked my GCode and everything is correct with the bed.  Something is going on though as I lowered my bed all the way down and ran the print again and the bed does not move at all.  Could it be some sort of issue with repetier not recognizing the callout for the bed?  I attached a printout of my gcode for a simple sample piece.  
[Sample Print.pdf](https://github.com/luc-github/Repetier-Firmware-0.92/files/920262/Sample.Print.pdf)

luc-github, 2017-04-13:
looks your acceleration for bed is little bit fast - may be too fast so it lose steps:
my settings:
[image]

which give: 
`G1 Z0.300 F600`

luc-github, 2017-04-13:
here my start gcode for your refrence :
```
G90
M82
M106 S0
M140 S110
M104 S0 T1
M104 S230 T0
M109 S230 T0
M116;wait all temperature to be reached
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 ; Home extruder
G1 Z15.0 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
M100
T0
G4 P2
;Indiana
M300 S1318 P240
M300 S0 P120
M300 S1396 P120
M300 S1567 P120
M300 S0 P120
M300 S2093 P960
M300 S0 P120
M300 S1174 P240
M300 S0 P120
M300 S1318 P120
M300 S1396 P960
M300 S0 P240
M300 S1567 P240
M300 S0 P120
M300 S880 P120
M300 S987 P120
M300 S0 P120
M300 S2793 P960
M300 S0 P240
M300 S880 P240
M300 S0 P120
M300 S987 P120
M300 S2093 P480
M300 S2349 P480
M300 S2637 P480
G92 E0
G1 E-2.0000 F1800
G1 Z0.300 F600
; layer 1, Z = 0.3
T0
; tool H0.300 W0.480
```

eriemaker, 2017-04-13:
Ok there is deffinately something going on with my setup, as I found out now it is not zeroing or going home before its start moving.  Therefore if my extruder is in the middle of the bed, when I start a print it is like it sets that as its home position and starts all of its moves from there.  Would that be in my gcode?

eriemaker, 2017-04-13:
By the way whats the Indiana and all the M, S, & P numbers?

luc-github, 2017-04-13:
music of indiana jones, I like fancy ;)

eriemaker, 2017-04-13:
lol, ok that makes sense!  On a good note, I took your starting script and put it in my gcode (including the indiana?) and it seemed to allow it to work.  It must be absolute positioning or homing in your code.  I will toy around with everything and see if there are any other issues.  I would like to reenable the decouple safety feature so I might toy with that.

eriemaker, 2017-04-13:
Also can you show me your ending script?

luc-github, 2017-04-13:
```
G91 ;relative positioning
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F150 ;move Z up a bit and retract filament even more
M100
G28 X0 Y0 ;move X/Y to min endstops  so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning
M104 S0 T0 ;extruder heater off
M104 S0 T1 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
;FunkyTown
M300 S2093 P240
M300 S2093 P240
M300 S932 P240
M300 S2093 P240
M300 S0 P240
M300 S1567 P240
M300 S0 P240
M300 S1567 P240
M300 S2093 P240
M300 S2793 P240
M300 S2637 P240
M300 S2093 P240
M300 S0 P960
```

eriemaker, 2017-04-13:
Thanks Luc!  one more question and hopefully I will be good.  I pulled my SD card out and reformatted it (for a random reason) and now when I upload .gcode files I cant find them from the printer screen.  Its almost like the firmware is not ignoring them.  Have you ever seen this?  I can see other files which is strange.

luc-github, 2017-04-14:
for formating SD I use : https://www.sdcard.org/downloads/formatter_4/

luc-github, 2017-04-14:
Ok so I close issue is all is OK now

closed 2017-04-14 by luc-github

patoinos, 2020-06-16:
buenas, mi extrusor derecho no funciona en repetier, de donde descargo el perfil para s3d

patoinos, 2020-06-16:
> Hola,
> estoy tratando de descubrir cómo ejecutar mi extrusor izquierdo solo en la configuración más repetida. Hice un intercambio donde puse un E3D hotend solo en el lado izquierdo, por lo que el lado derecho no tiene extrusora. Estaba recibiendo un error de desacoplamiento del calentador que solucioné y está ocurriendo porque el extrusor correcto no está allí. Encontré el problema [# 94](https://github.com/luc-github/Repetier-Firmware-4-Davinci/issues/94) , donde Luc dijo esto: "está bien, sí, es un caso de esquina, por lo que su solución para establecer NUM_EXTRUDER en 1 resolverá el problema, pero puede ocasionar problemas en otros lugares".
> 
> Pensé que eso solucionaría mi problema, pero luego me di cuenta de que cuando lo hacía, se aplicaba todo al extrusor correcto. ¿Hay una solución simple que pueda hacer en el archivo de configuración que cambiaría todo para decirle que solo tengo un extrusor izquierdo? Uno de mis pensamientos era simplemente conectar los cables del extrusor, hotend, etc. de derecha a izquierda, pero luego necesitaré algún tipo de desplazamiento x, ya que pensará que se imprime desde el lado derecho. O puedo separar la impresora nuevamente y colocar el extrusor en el lado derecho, aunque me gustaría evitarlo.
> 
> Para lo que vale, deshabilité la alerta de desacoplamiento y cambié mi configuración de simplificación para imprimir desde el extrusor izquierdo, pero cuando se calentó recibí una alerta para cargar el filamento en el extrusor derecho.

alguien tiene la solucion para usar el segundo extrusor en la davinci duo con repetier

luc-github, 2020-06-16:
@patoinos porque dices que non functiona? Tengo un Davinci Duo y no tengo nigum problemo
 S3D y Repetier son differentes, el slicer es differente tambien repetier puede usar cura y slic3r y tienes que modifiar el start  GCODE para ellos

Repetier : 
Derecho extrusor E1 
Izquierda E0

lo siento, hace anos que no yo he hablado/escribido espanol...

patoinos, 2020-06-16:
Lo digo, por que al seleccionar el extrusor 2 no hace nada y automáticamente usa el 1, 

 tienes que modificar el start GCODE para ellos

Repetidor: 
Derecho extrusor E1 
Izquierda E0

Cómo se hace eso

luc-github, 2020-06-16:
que slicer estas usando ?

patoinos, 2020-06-16:
hola ! gracias por el tiempo estoy usando curaengine del repetier host

luc-github, 2020-06-16:
Mira aqui : solo hay por E0 / T0 (izqueria) y nada para por E1/T1 (dercha)
necessitas las lineas para E1/T1
[image]

luc-github, 2020-06-16:
todo este no es specifico para davinci

patoinos, 2020-06-16:
genial !!! te pasaste y solo debo agregar esos valores ? muchas gracias amigo ! me llamo patricio de verdad muchas gracias llevo semanas sin poder usarlo