luc-github/Repetier-Firmware-4-Davinci #239: Davinci 1.0/J37/Black Bars/USB Not recognized/Bootloop?/

Pistow, 2017-04-15:
Hi

To start with I own a Davinci 1.0 with J37 and micro sd card. My os is win10 pro 64bit.

Showed up as Bossa in device manager, I was so happy until I fucked up and tried to flash my printer with IDE 1.8.2 and due 1.6.11 ... (yeah I know im retarded)

So now the printer is stuck with those black lines and windows says usb non recognized in loop like printer was on then off then on etc (no com port ==> can't reflash)
In device manager, device shows code 43 which stands for ["One of the drivers controlling the device notified the operating system that the device failed in some manner."](https://support.microsoft.com/en-us/help/310123/error-codes-in-device-manager-in-windows)

Seems like jumping j37 doesnt change anything, tried rebooting printer/pc, using a different PC with win10 (usb non recognized) and a macbook (didn't even show up in IDE) and different usb ports but nothing. I also tried reinstalling IDE (1.8.0/DUE 1.6.8), following the "tutorial" printer still shows as non recognized. Uninstalling drivers in device manager isn't working too Tried connecting printer to pc with another usb cable but still not recognized 

I can backup my computer to April 4th if that can help but it seems like it's an hardware issue ?
Haven't touched the board except the J37 jumpers so I don't know what I could have damaged on the board itself

Hope I didn't permanently bricked my board and that this issue is solvable
Thanks in advance !

luc-github, 2017-04-15:
Wiping flash with Jumper should allow to detect printer as bossa or due when you did not do the proper variant.cpp update

Be sure you wipe slowly : 
1- power down printer 
2 - jump the contacts
3 - power on with contact jumped
4 - wait for 10s still with contacts jumped
5 - power Off still with contacts jumped
For safety I always do a second time
6 - power on  again with contacts jumped
7 - wait for 10s still with contacts jumped
8 - power Off still with contacts jumped

if doing this the board is not recognized it means there is some hardware defect

I would also add I saw some people jumping the wrong jumper and fry the board so be cautious

Pistow, 2017-04-15:
Just did it and board is still in "bootloop" / not recognized
I don't know if that's important but the 3 leds on the back are on

Now if the board is really dead, what should I do ? What board can I buy to replace it or should I contact XYZ Support ? I believe I'm still under warranty.
[image]
[image]

luc-github, 2017-04-15:
if you still under warranty - yes You can contact support and repport one morning you power on the printer and got this.
Do not mention anything else if I may suggest

You can try also some forums to buy second hand boards  as this board is not on sale alone as far I know

just flashing should not brick the printer - also remember you must install  arduino 1.8.0 / due 1.6.8 on every computer you try to connect your printer to get ther proper drivers

Pistow, 2017-04-15:
So there's really nothing I can try :/ ?
Guess I'll have to contact XYZ then

luc-github, 2017-04-15:
Sorry - electronic is not my part and wiping with jumper means only electronic left - Firmware is no more present, so...

Pistow, 2017-04-15:
Alright thanks a lot anyway :/ !

closed 2017-04-15 by Pistow