luc-github/Repetier-Firmware-4-Davinci #241: Davinci 2.0 Duo Home is messed up!

FickleMillipede, 2017-04-17:
After a few weeks I FINALLY got this firmware to work on my Da Vinci 2.0 Duo... only problem is that the home is set so that the left extruder is jammed against the right side of the print bed, and i have no idea how to fix it. When i flashed the firmware I didn't realize that the bed was all the way up and that the home would set there. I cant find a way to reset the home positions to the normal printer positions. I also would like to know what the settings would be for it, if you know?
Thanks

luc-github, 2017-04-17:
I do not understand the issue you meet
what means :  
>I cant find a way to reset the home positions to the normal printer positions.

if you load fail-safe EEPROM settings or send M502 then M500 commands a G28 will do the homing

putting extruder on right/back position.
Extruder left is T0, first extruder = default one, Extruder right is T1 second extruder

luc-github, 2017-04-22:
any update on this ?

FickleMillipede, 2017-04-22:
Sorry, I got a little busy. Okay, the first problem I mentioned above has been sortve fixed. The problem now is that when I manually calibrate the bed, it's way too high, the bed is down all the way and the extruders almost punch a hole through the bed. Also, when I start a print, the extruder and bed home after they've finished heating, but when it starts to print, it grinds the extruder head on the side of the bed for about a second, then the bed lowers down by about 5mm and it prints in the air.... do you have any idea what it could be? Did I just get a bad printer? Thanks for your help

luc-github, 2017-04-22:
the bed is too high when you manually calibrate ?
Be sure your Z offset and Z min is set to 0. - when you home Z your extruder should be in correct position for 0, if not it means you do not do proper bed leveling 

the printer only follow GCode you send, so if printing is not in correct position it means Gcode is not correct  due to wrong slicer setting  

there is no bad printer only bad settings

luc-github, 2017-05-04:
I close issue as no update

closed 2017-05-04 by luc-github