luc-github/Repetier-Firmware-4-Davinci #242: Can't see Com port after shorting J37

exxtreme5, 2017-04-22:
hi there,
I have reflash my Da Vinci 1.0a a couple of times before. However on the last try when I jumped my J37 to reset the chip, powering up the printer seems ok, but the computer doesn't seem to see the com port anymore. Have tried changing the usb cable and used another computer but no avail. Have also tried to reshort J37 a couple more times but still nothing. Is my board fried?

Thanks!

luc-github, 2017-04-22:
after shorting the Jumper there is no FW so board should be detected
even you use jumper every time you flash (which is not necessary) you should not damage the board 

so somehow the board is damaged or the usb port is damaged - I cannot say more sorry
Davinci is very fragile - you may have accidentaly short the wrong jumper ? or did a short cleaning extruder ? many ways to damage board accidentaly

exxtreme5, 2017-04-22:
Sad i think something is wrong with the board somehow. I have reflash the firmware at least 5 times as I was added in a z-probe.
Going to swap it out for RAMPS, not keen on all the work required for this =(

Thanks for the hardwork, the fw work like a charm on my 1.0A other than this mishap.

luc-github, 2017-04-22:
Sorry for you - this is a small task to swap - furtunaly several forum document it 
May I ask you if you think it worth to port the davinci fork to ramps/mega ? or using default repetier/marlin is enough ?

luc-github, 2017-05-04:
I close isssue if no update

closed 2017-05-04 by luc-github