luc-github/Repetier-Firmware-4-Davinci #243: Extruder1 has no filament - Davinci 2.0A

steve89corrales, 2017-04-26:
Whenever I try to print, with either extruder, after the extruders heat up, I get that fault.  Both extruders have filament.  I have replaced the extruder and the problem remains.  It really seems to be an issue in the firmware where it does not acknowledge filament in the left hand extruder.  When I try to unload that extruder, once it reaches temp, it only runs for a short period of time and then moves back to the right.  I see that as the extruder does not sense filament and is moving back to the home position.

I have flashed the firmware multiple times as both 2.0 and 2.0A (it is a 2.0).

Any help would be wonderful.  Thank you.

Steve

luc-github, 2017-04-26:
The logic and code of filament sensor did not changed since a long time so I doubt the issue in firmware

the filament sensor is actually an optical sensor that need to be covered to detect filament.
so it detect no filament: 
- when nothing block the optical sensor, 
- when filament is a clear filament that does not block light fully (any clear filament), 
- when sensor is damaged, sensor wire is cuted or sensor is missing like if daugther board is removed, the signal is not blocked.

You are not is first case as you say both extruders have filament,  so you should be in one of the 2 others cases.
In all cases the workaround is to go to settings and set Fil.Sensors: Off to disable sensor

steve89corrales, 2017-04-26:
luc-github, thank you.  Turning off the sensors is now allowing me to print.  I can now look into the sensor problem at a later time.

closed 2017-04-26 by steve89corrales