luc-github/Repetier-Firmware-4-Davinci #245: how to install custom firmware on Da Vinci Jr.

prahalad1224, 2017-05-03:
I have tried using ATSAM-BA 2.1 for installing custom firmware but after upload board not detected.
if anyone has binary file of firmware please share

luc-github, 2017-05-03:
There is no FW for JR currently - did you read the readme ?

 there are some work on going but not using arduino and not based on repetier FW and currently not workable

prahalad1224, 2017-05-03:
[3F1J0_FW_V2.2.0.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/972961/3F1J0_FW_V2.2.0.zip)

This the FW on which Im working. but after installation, usb port gets disabled.

prahalad1224, 2017-05-03:
[Duet-WiFi.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/972969/Duet-WiFi.zip)

and this the wifi FW which I think is compatible with Jr.
help pls what to do?

luc-github, 2017-05-03:
to flash Jr with duet FW you need to use samba as currently I did not finished to work on bossac for sam4e
I use 2.16
[image]

be aware duet FW is absolutly not compatible with JR - only the core but not the pinout and I am 99% sure the variant need to be modified as I was not able to find the heater and thermistor pin

Did you worked on jr pinout ?

prahalad1224, 2017-05-03:
[Jr. pinouts.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/973123/Jr.pinouts.xlsx)

What do you mean by only the core?
and yes i have pinout list and have tried it also.
but dont know where Im lacking knowledge.
and yes it is variant.cpp
If I give you pinout list can you share me the modified variant.cpp?

pinout list attched here.

luc-github, 2017-05-03:
this is not the pinout just the pin definition from MCU point of view

here the current know pinout that I have shared :
http://www.soliforum.com/post/131637/#p131637
we have end stop / light / motors pins
still miss heater pin, thermistor pin, sdcard reader pins, lcd pins, filament sensor pin, Jam sensor pin, others TBD ?

also duetFW does not support lcd / keypad so need to add it once pins are found

luc-github, 2017-05-03:
also be aware that after erasing FW - the extruder is heating without control as soon as printer is power on so better to disconnect it when doing your tests

prahalad1224, 2017-05-03:
ok thank you for your support.
Can you please share me the duet .bin  file with details procedure which you have modified to use in SAMBA 2.1?

and meanwhile I will see the pinouts for heater and thermistor.

prahalad1224, 2017-05-03:
also, please share modified variant.cpp on my email.
prahalad.solanki.90@gmail.com

it will be a great help.
thank you

luc-github, 2017-05-03:
Sorry I have nothing to share as I did not modified the vaiant yet - I am currently working with existing variant
I did not modified the duet FW for JR, I did some test FW to test find / the pins but nothing for the printer itself

luc-github, 2017-05-03:
if you can find pin for extruder and thermistor would be great 
I am currently busy with another project but will come back to JR porting soon

luc-github, 2017-06-19:
I close issue then - feel free to reopen if needed

closed 2017-06-19 by luc-github

the-imperfectionest, 2018-12-08:
I have one of these on loan to "fix" for a friend.  Whats the best way to get the pinouts for you?

luc-github, 2018-12-08:
Actually the pinout is now known - (http://www.soliforum.com/topic/13769/davinci-jr-repetier-port/page/6/)  the reason why thermistor / heater what not detected was because they are handled by another mcu which make porting even more complex as there are 2 solutions :
1 - hardware rework to re route the pins to main MCU - not easy for most of people
2 - reprogram the MCU in charge of temperature - LPC1115  is an NXP mcu -  ideal case would be to be able to  flash LPC1115  using the main mcu - not sure it is possible, another solution whould be to use the header to flash it but this need soldering - and this one it is not easy

So it become harder to hack but still not impossible