luc-github/Repetier-Firmware-4-Davinci #246: 1.0A New Flash Issues

sammes19, 2017-05-07:
Windows 10 64Bit, Arduino 1.8.0, with 1.6.8 Mod for Due 

I flashed the printer followed along the steps listed in Youtube Videos as well as on this page. I have successful compile and upload. After turning off and back on, I still have the black bars across the LCD and the my windows laptop is telling me unrecognizable device. I was able to follow another method and install .91 on it, but it does not function correctly since this is the 1.0A model. I am new to this entire process so if someone could help me figure out where the issue could be occurring. Also how can I get simplify3D to work, do I load the typical setup for the davinci, and go from there?

luc-github, 2017-05-08:
Did you check the FAQ ? black bar is typical issue

You can give a try also on prebuilt available here https://github.com/luc-github/Repetier-Firmware-0.92/releases using bossac directly

sammes19, 2017-05-08:
I have followed the FAQ, and notice the issue typically arises from incorrect variant.cpp and USB. The issue i was having was I could not locate the files in Arduino 1.8.0 through program files (x86) on my computer. I have several folders loaded within Hardware>Variants, with no direct file of variant.cpp. I will try the second link you sent me, that way worked for me loading .91. Should I still focus on using Arduino 1.8.0, and take bossac from there to load it? And then use the .bin for my specific model printer. Thanks in Advance

luc-github, 2017-05-08:
So if you read FAQ you saw there is a FAQ for finding variant.cpp : https://github.com/luc-github/Repetier-Firmware-0.92/issues/114

sammes19, 2017-05-08:
Correct, that is the exact one I referenced, but I have different folders than what is being listed there.

luc-github, 2017-05-08:
this FAQ explain how to get right path which vary depending on OS and ide version and due module version
Following the properly the readme it must be for window :
`%USERPROFILE%\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x`

May I know what you do to find the path ? 
or can you share what it is not clear in this FAQ  to find the path start `%USERPROFILE%\AppData\Local\Arduino15` using preferences dialog ?
I am curious

sammes19, 2017-05-08:
If I use AutoBuild Binary, What version of Arduino should I download? Where do I access Bossac for the version of Arduino?

luc-github, 2017-05-08:
on ide 1.8.0 you can find bossac in same way as variant.cpp but instead of hardware it is in tools
%USERPROFILE%\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino

Anyway as I wrote before, any bossac version will work 
so just do same you did for 0.91 but use new binary FW

sammes19, 2017-05-08:
I was able to locate the files through the pathway you have shown, Everything is working. Really appreciate the time you have spent on this!

sammes19, 2017-05-08:
How Do I go about connecting this to simplify3D, I get connection fail through the USB

luc-github, 2017-05-08:
like any repetier fw setting in fw configuration of S3D
select check  hardware flow control , unckeck wait for startup command start, etc...

sammes19, 2017-05-08:
It has connected now by changing the flow to hardware, and uncheck wait for startup command. I have now tried uploading via USB and it says .3W files can only be uploaded to Xyz machines, any fix for this. If you could please explain. I am new to this. Thanks

sammes19, 2017-05-08:
Also, there is no way to disconnect my computer and run the printer stand alone with this software? How can I upload directly onto SD card with G-Code like you have already listed through print file. Again, I am using S3D, and New to this so please explain if you can.

luc-github, 2017-05-09:
Issue is solved - I close ticket 
Because this is a bug tracker for repetier for davinci printer,  I suggest you to check S3D documentation for everything related to S3D
using google you have :   https://forum.simplify3d.com/ and https://www.simplify3d.com/support/
https://3dprinting.com/what-is-3d-printing/

Wish you good luck in your learning curve

closed 2017-05-09 by luc-github