luc-github/Repetier-Firmware-4-Davinci #247: what is the easy way to upgrade to the latest version of Davinci Repetier

celticht32, 2017-05-08:
I am using a version of Repetier Firmware that is about a year old maybe two.
I have noticed that since I upgraded to the latest version of Repetier Server there seems to
be some weirdness happening.... Such as the server has to be restarted sometimes in order to get
the server to reconnect with the printer.... if I dont reboot the printer and the server after a few prints
the prints start shifting to the right..... rebooting both resets the position. So I am thinking it might be
the fact that I am running an out of date version of firmware.
So what is the easy way to upgrade the firmware.... do I have to go through the same process as when I 
reflashed? Or do I just connect up Arduino IDE and compile and push?

luc-github, 2017-05-08:
to upgrade you do not need to use Jumper that is correct

How to upgrade will depend of version of ide you installed - latest FW is only compatible with ide 1.62+ (I think, but not sure what is the exact version) - if compilation pass it means it is ok 
be aware if you update or reinstall ide or due module, you must redo the variant.cpp update or you go to the classic black bars problem

I do not think you need to update FW (unless you really want) - I guess issue is more on Repetier server itself, so unless you really need latest version, or it is etablished it fw version incompatibility ( you can find this in release note of repetier server, sorry I do not use it so I do not follow, so I am not aware of such thing)  - it is better to revert to working version of repetier server

Keep us updated

luc-github, 2017-05-22:
as no update I close issue

closed 2017-05-22 by luc-github